silvanmelchior/RPi_Cam_Web_Interface #513: raspimjpeg: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found

GEJENL, 2019-08-28:
Guys, 
Nice project, still running on another pi, now i am trying a new install, but that fails with above message...

Installed clean raspberry stretch, which comes with glibc-2.24, but you need 2.28?

Question: how can i upgrade to that version? 
See my attempts below. (newby, sorry)
thanks!
Gert-Jan

pi@GJ-PI-BEL:~/RPi_Cam_Web_Interface $ sudo apt-get install libc6
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libc6 is already the newest version (2.24-11+deb9u4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@GJ-PI-BEL:~/RPi_Cam_Web_Interface $ sudo apt-get install libc6=2.28
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Version '2.28' for 'libc6' was not found

GEJENL, 2019-08-28:
> The package doesn't explicitly install a particular version of gclib but I suspect this may be dependency of the php version.
>
> Latest install has defaults appropriate for Buster rather than Stretch so expects to install php 7.3
>
> For Stretch You should change this to 7.1 during the install start up.
>
> I would, however, recommend starting from clean Buster now.

Thanks for your swift answer!
Impressed... -:)

Downloading buster now.

Gert-Jan

roberttidey, 2019-08-28:
The package doesn't explicitly install a particular version of gclib  but I suspect this may be dependency of the php version.

Latest install has defaults appropriate for Buster rather than Stretch so expects to install php 7.3

For Stretch You should change this to 7.1 during the install start up.

I would, however, recommend starting from clean Buster now.

GEJENL, 2019-08-28:
Thanks! 
Still time to start allover with 'buster' :-)

closed 2019-08-28 by GEJENL