silvanmelchior/RPi_Cam_Web_Interface #514: Macros and scheduler, small query

35nine, 2019-08-30:
Hi, thanks for this continuing software, I am running still on 6.5.1.  I have macros set for beginning of the night period (use sun based time bands).  For example macro.sh.  It fires off at the right time, for another example, 20:00 in logs sy macro.sh etc.  Perfect.

When the next day begins, later that night, at 00:00 or 00:01, the macro fires again.  In the log I have sy macro.sh etc and external verification via a HA dashboard that it did trigger.  

I am wondering if this is normal or something that can be thought if possible to change?  Can it be the macro can be set to fire at sunset as needed then not when the day ticks into the next day?  Hope this makes sense, cheers for any thoughts

roberttidey, 2019-08-30:
Macros configured should only fire when the period of the day changes.

Could you post here the section of the log and the cschedule.json (from /var/www/html or equivalent) so I can investigate further.

35nine, 2019-08-30:
Hi, shedule attached (I think) and snip from log.

Appreciate the help!

[schedule.zip](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/files/3561708/schedule.zip)

[2019/08/30 22:52:44] Scheduled management tasks. Next at 1567173164
[2019/08/30 23:52:45] Scheduled management tasks. Next at 1567176765
[2019/08/31 00:00:02] New period detected 1
[2019/08/31 00:00:02] Send em night
{2019/08/31 00:00:02} Change: exposure_mode = night
[2019/08/31 00:00:04] Send sy ledoff.sh
{2019/08/31 00:00:04} Executing macro /var/www/html/macros/ledoff.sh &
[2019/08/31 00:00:06] Send sy redclose.sh
{2019/08/31 00:00:06} Executing macro /var/www/html/macros/redclose.sh &
[2019/08/31 00:52:46] Scheduled management tasks. Next at 1567180366
[2019/08/31 01:52:47] Scheduled management tasks. Next at 1567183967
[2019/08/31 02:52:48] Scheduled management tasks. Next at 1567187568
[2019/08/31 03:43:09] New period detected 2
[2019/08/31 03:43:09] Send em night
{2019/08/31 03:43:09} Change: exposure_mode = night
[2019/08/31 03:52:49] Scheduled management tasks. Next at 1567191169

roberttidey, 2019-08-31:
Thanks for json and snip of log.

II think I can see why this is happening. It is due to a change made to fix an issue with the fixed time periods.

I should be able to resolve this soon.