silvanmelchior/RPi_Cam_Web_Interface #516: MP4 boxing never starts

Pi-DD, 2019-08-30:
Have fallen back to 6.5.1.

Video is captured when motion occurs, file added to queue, but boxing never starts. GPAC is installed, I have checked.

Where is end_vid macro so I can check it? Does not appear in /var/www/html/macros or equivalent RPi_Web_Cam_Interface directory

roberttidey, 2019-08-30:
end_vid macro is for user extensions and does not need to exist for normal operation.

Your problem with Boxing may be due to your failed upgrade and then attempt to revert.

If possible I would suggest a clean install starting from Buster

Pi-DD, 2019-08-31:
Buster update ongoing now as per other issue. 

Can you suggest a way of boxing the h264 files? 

I tried running MP4box but got a message that it did not like the input file.

roberttidey, 2019-08-31:
MP4Box is the standard way. What command line did you use?

This software runs
`MP4Box -fps %i -add %s %s`

where %i is substituted by the frame rate e.g 25, the first %s is the input file (full path like /var/www/html/media/myvid.h264 and the second %s is the full path to destination file like /var/www/html/media/myvid.mp4

Pi-DD, 2019-08-31:
Thanks Robert - I suspect frame rate was my error! I’ll try again anon.

Pi-DD, 2019-08-31:
Have now updated to 6.6.4 and Buster. 

Got the following at end of running update.sh

`pi@DDhomepi3:~/RPi_Cam_Web_Interface $ cPHP Notice:  A non well formed numeric value encountered in /var/www/html/schedule.php on line 650
d  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   121  100    54  100    67     88    109 --:--:-- --:--:-- --:--:--   197
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   121  100    54  100    67     72     89 --:--:-- --:--:-- --:--:--   162
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   121  100    54  100    67     73     90 --:--:-- --:--:-- --:--:--   164
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   121  100    54  100    67     65     81 --:--:-- --:--:-- --:--:--   146
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   121  100    54  100    67     95    118 --:--:-- --:--:-- --:--:--   213
`
Some older captures were then boxed but not all, so deleted them 

Still getting same problem. Motion is detected, file captured, but never boxed and left marked “busy”.

Pi-DD, 2019-08-31:
Boxing problem Solved! Turns out that the residual h264 files block boxing from occurring.
When I deleted the, suddenly got the flow of IFTTT notifications I have set up using end_box.sh for those I deleted and subsequent detections work and create MP4 files ok. 

Closing this and opening separate issue for message after update

closed 2019-08-31 by Pi-DD