silvanmelchior/RPi_Cam_Web_Interface #518: libc.so.6 version 'GLIBC_2.28' not found

florid2, 2019-09-02:
I experienced couple issues with installation. 
I tried to customize the web port from 80 to other port so that I have run pihole on the same pi. But failed existing RPi Cam project and have no way to get it back working. 
I tried remove.sh, apt autoremove and apt purge apache2, lighttpd. Hope I can get a clean system as possible. And of course reboot. 
My pi 3B is running latest raspbian stretch. 

I cloned the RPI git then run the ./install.sh. 
only modify the web port to something like 8880, and username/password, php7.0. The installation script runs fine. Only complains /lib/arm-linux-gnueabihf/libc.so.6 version 'GLIBC_2.28' not found required by raspimjpeg after start the service. 

When I login the web portal, there's no realtime image displayed, but shows loading. 

I also tried default installation, leave port 80 and apache2, username/password set and php7.0. The same issue of GLIBC not found and web portal not display footage. 

And clue how to troubleshoot this? Thanks.

florid2, 2019-09-02:
I may figure out that the install script is no longer support stretch but only buster. 
But your installation script is still confusing that php7.0 (stretch), php7.3(buster). If the installation is intended not support stretch, you should remove that option.

roberttidey, 2019-09-02:
I do recommend using Buster now.

The installation script is intended to support Stretch as well but you do then have to change the python version in the configuration menu to 7.0 as Stretch doesn't support 7.3.

It is also possible that compiling raspimjpeg under Buster has introduced an extra dependency. I'll have to check into that.

visser, 2019-09-19:
https://elinux.org/RPi-Cam-Web-Interface#Basic_Installation

For preBuster OS you will need a different version of raspimjpeg in the bin folder. Rename raspimjpeg to raspimjpeg-buster and raspimjpeg-stretch to raspimjpeg before installing.

(Noticed I'm running Stretch due to PiBakery not supporting Buster, applied the filename change prior to running ./install and all is working)