silvanmelchior/RPi_Cam_Web_Interface #521: unable to use audio recording

rajeshkrarora1, 2019-09-26:
Hi there, i installed from https://github.com/konstantinj/RPi_Cam_Web_Interface, installed RPi_Cam_Web_Interface using apache server, attached usb mic. The camera works fine but am unable to record audio with video, there are no details as in how to configure the software to work with audio. need help Thanks in anticipation

Thanks and regards

ffcruz85, 2019-09-30:
pretty sure there are no audio recording capabilities here

roberttidey, 2019-09-30:
That is correct. The software as supplied does not support audio. It is possible to add separate audio recording by using the macro capabilities. E.g. initiate an audio recording via the start_vid and stop it via end_vid.

ffcruz85, 2020-12-10:
Hey @roberttidey - I know this is old, but do you have any working examples of that? I guess it's easy enough to just start a recording, but just curious if you have seen any examples of something a little more sophisticated, like naming the file the same as the video file name, or more importantly, automatically applying the recorded audio to the video itself?
I have something in mind for the first parts, some logic like get list of files in directory, sort by newest with vi_ in the name, and save audio file with that name. But I have no idea where to begin with merging the audio into the video file. Sorry, I know it's not supported and I am not requesting a feature, just figured you might have run across an example at some point before I start trying to reinvent the wheel/
THanks for all the good work - I love your tool and have been using it for years.

ffcruz85, 2020-12-10:
actually, another idea - I could use the motion event to start a macro but not start a capture from RPi_Cam_Web_Interface, calling instead for another program to do the recording with audio. two potential issues there:
1) Would this create a conflict with RPi_Cam_Web_Interface since it is streaming, with the other program being blocked from using the camera?
2) The video would not show up in the preview.php page - other than saving the video file in www/media with the correct naming pattern, what would I have to do for it to show up in the list?

roberttidey, 2020-12-11:
I personally don't use audio as my cameras are primarily for security and don't have microphones.

I know some users have used macros to kick off audio recordings whenever video recordings start and then combined them afterwards but I haven't been given any details. I did also offer to integrate audio facilities into raspimjpeg ( used by RPi_Cam_Web_Interface) if someone wanted to help with that but that didn't progress.

On the idea of just using the motion detection to kick of a separate combined/audio video recording then one would need to be careful to workaround camera interface conflicts. With the RPi_Cam_Web_Interface streaming and running motion capture another application cannot access the camera. However, when motion capture triggers a start one could suspend the RPi_Cam_Web_Interface from using the camera (cmd ru 0), do the separate recording, then restart the RPi_Cam_Web_Interface camera (cmd ru 1) after the recording is complete. Of course one can't use motion detection to sense lack of movement to stop the recording so one would need to do something like fixed time duration recordings.

For items to show up on preview the key is actually the thumbnails. These are what is scanned to determine what is there. Thumbnails are jpeg files which always end with .xnnnn.th.jpg where x is the type of main data file {v(video), i(image) or t(timelapse)} The main data file is then the same name but without that ending. If you ended up with a separate mp4 recording then you could just attach that type of ending to name say a generic jpg image to be used as a thumbnail.  There are functions in the config.php which determine what is a thumbnail and then to extract its type, index number and to return the original data filename.

sbkomurcu, 2020-12-14:
I am currently working on this and I am now able to start/stop recording via macros.
In order to do this, I have edited the raspimjpeg file where the conversion command takes place, (see [line 87](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/blob/master/etc/raspimjpeg/raspimjpeg.1), so that I am importing the `mp3`  file into the conversion process and I get a video with audio.
 
However, I am still working on it as the audio-video sync is not perfect. I will publish the work I have done once it is usable.

roberttidey, 2020-12-14:
Excellent. I look forward to the result. 

Sync should be easier to achieve if no extra video buffering is used as that introduces extra pre trigger offset at the start of the video.

wrapper, 2021-06-21:
> I am currently working on this and I am now able to start/stop recording via macros.
> In order to do this, I have edited the raspimjpeg configuration file where the conversion command takes place, (see [line 87](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/blob/9fb4aed4bf66ef148d667ea648f757aa2e57bf2a/etc/raspimjpeg/raspimjpeg.1#L87)), so that I am importing the `mp3` file into the conversion process and I get a video with audio.
> (Please note that I have edited the raspimjpeg file after the installation, It can be found at /etc/raspimjpeg)
> 
> However, I am still working on it as the audio-video sync is not perfect. I will publish the work I have done once it is usable.

Have you made any progress? Thanks

sbkomurcu, 2021-06-21:
> Have you made any progress? Thanks

Yes, I am going to publish a gist regarding what I made. It is gonna be ready latest by the end of this week.

wrapper, 2021-07-14:
> Yes, I am going to publish a gist regarding what I made. It is gonna be ready latest by the end of this week.

Great! Where can I find it? Thanks

chadhutchins182, 2022-08-16:
@sbkomurcu Can we get the gist please?

sbkomurcu, 2022-08-24:
Hey @chadhutchins182. It's already published by someone else before I publish. Please refer #634. The solution I mention was the same as in the #634. Simply add macro for start/stop audio recording (using end_vid, start_vid) and edit the boxing behaviour as mentioned in the post.

There is one thing I did different is that the the recording file names were exactly same as the video file name but with different file extension.  (i.e 2022-12-10.h264 - 2022-12-10.mp3 --> 2022-12-10.mp4) Therefore every video file have its own unique audio file. 
Sorry for the late answer, somehow I haven't see the notifications.