silvanmelchior/RPi_Cam_Web_Interface #523: ca 1 [t] command: unexpected behavior when Video Split is nonzero

ras07, 2019-10-14:
If the **Video Split** interval is set (non-zero) and you issue (for example) **ca 1 20** to FIFO11, the system records 20 seconds worth of video as expected, but then starts recording again and runs (as far as I can tell) forever. Log:
`{2019/10/13 18:53:16} Capturing 20 seconds
{2019/10/13 18:53:16} Executing macro /var/www/html/macros/start_vid.sh > /var/www/html/macros/start_vid.sh.out 2>&1 "/var/www/html/media/vi_00020_20191013_185316.mp4" &
{2019/10/13 18:53:16} Capturing started
{2019/10/13 18:53:36} Stopping video from timer
{2019/10/13 18:53:37} Capturing stopped
{2019/10/13 18:53:37} Add /var/www/html/media/vi_00020_20191013_185316.mp4 to Box Queue at pos 22
{2019/10/13 18:53:37} DEBUG 3
{2019/10/13 18:53:37} Restarting next split of 300 seconds
{2019/10/13 18:53:37} Executing macro /var/www/html/macros/start_vid.sh > /var/www/html/macros/start_vid.sh.out 2>&1 "/var/www/html/media/vi_00021_20191013_185337.mp4" &
{2019/10/13 18:53:37} Capturing started
{2019/10/13 18:53:38} Start boxing /var/www/html/media/vi_00020_20191013_185316.h264 to /var/www/html/media/vi_00020_20191013_185316.mp4 Queue pos 22
{2019/10/13 18:53:45} Finished boxing /var/www/html/media/vi_00020_20191013_185316.mp4 from Box Queue at pos 22
{2019/10/13 18:53:45} Executing macro /var/www/html/macros/end_box.sh > /var/www/html/macros/end_box.sh.out 2>&1 "/var/www/html/media/vi_00020_20191013_185316.mp4" &
{2019/10/13 18:53:45} Removed item from Box Queue
{2019/10/13 18:58:37} Stopping video from timer
{2019/10/13 18:58:37} Capturing stopped
{2019/10/13 18:58:37} Add /var/www/html/media/vi_00021_20191013_185337.mp4 to Box Queue at pos 23
{2019/10/13 18:58:37} DEBUG 3
{2019/10/13 18:58:37} Restarting next split of 300 seconds
{2019/10/13 18:58:37} Executing macro /var/www/html/macros/start_vid.sh > /var/www/html/macros/start_vid.sh.out 2>&1 "/var/www/html/media/vi_00022_20191013_185837.mp4" &
{2019/10/13 18:58:37} Capturing started
{2019/10/13 18:58:38} Start boxing /var/www/html/media/vi_00021_20191013_185337.h264 to /var/www/html/media/vi_00021_20191013_185337.mp4 Queue pos 23`
etc.
If you zero out the **Video Split** setting, it behaves as expected.

roberttidey, 2019-10-14:
Yes. The video split was introduced for users who were doing continuous recording and wanted the video set into shorter segments.

Currently if it is set then the fixed time recording set by the optional second second parameter of ca is overridden and continuous recording is performed.

I'll  check into changing this behaviour.

roberttidey, 2019-10-14:
Latest version (6.6.6) now allows time limited capture (ca 1 n) with video split and will terminate after n.

ras07, 2019-10-14:
Thanks! Closing.

closed 2019-10-14 by ras07