silvanmelchior/RPi_Cam_Web_Interface #525: dependency issues with buster

7176462, 2019-10-23:
Hi. I'm new to pi and trying to follow the instructions on your elinux.org/RPi-Cam-Web-Interace page.  Sorry such a long email and any other inconveniences.  I have a pi 3 model b+ using a pi camera v2.  When I get to Step 4 and ./install.sh I get the following:
Building dependency tree       
Reading state information... Done
zip is already the newest version (3.0-11).
zip set to manually installed.
motion is already the newest version (4.2.2-1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 apache2 : Depends: apache2-bin (= 2.4.38-3) but it is not going to be installed
           Depends: apache2-data (= 2.4.38-3) but it is not going to be installed
           Depends: apache2-utils (= 2.4.38-3) but it is not going to be installed
           Recommends: ssl-cert but it is not going to be installed
 gpac : Depends: gpac-modules-base (= 0.5.2-426-gc5ad4e4+dfsg5-5) but it is not going to be installed
        Depends: libgpac4 (= 0.5.2-426-gc5ad4e4+dfsg5-5) but it is not going to be installed
 libapache2-mod-php7.3 : Depends: php7.3-common (= 7.3.4-2) but it is not going to be installed
                         Depends: php7.3-json but it is not going to be installed
                         Depends: php7.3-opcache but it is not going to be installed
                         Depends: apache2-api-20120211
                         Depends: apache2-bin (>= 2.4.16) but it is not going to be installed
 motion : Depends: libmariadb3 (>= 3.0.0) but it is not going to be installed
          Depends: libmicrohttpd12 (>= 0.9.50) but it is not going to be installed
 php7.3 : Depends: php7.3-common but it is not going to be installed
 php7.3-cli : Depends: php7.3-common (= 7.3.4-2) but it is not going to be installed
              Depends: php7.3-json but it is not going to be installed
              Depends: php7.3-opcache but it is not going to be installed
              Depends: php7.3-readline but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

Then when I fix broken install:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  libmariadb3 libmicrohttpd12 mariadb-common mysql-common
The following NEW packages will be installed:
  libmariadb3 libmicrohttpd12 mariadb-common mysql-common
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 198 kB/292 kB of archives.
After this operation, 715 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf mariadb-common all 1:10.3.15-1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf libmariadb3 armhf 1:10.3.15-1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/m/mariadb-10.3/mariadb-common_10.3.15-1_all.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/m/mariadb-10.3/libmariadb3_10.3.15-1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

After which I tried ./install.sh and I get blue screen with apache as the server and php 7.3 and I choose OK.  I get this:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
zip is already the newest version (3.0-11).
zip set to manually installed.
motion is already the newest version (4.2.2-1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 apache2 : Depends: apache2-bin (= 2.4.38-3) but it is not going to be installed
           Depends: apache2-data (= 2.4.38-3) but it is not going to be installed
           Depends: apache2-utils (= 2.4.38-3) but it is not going to be installed
           Recommends: ssl-cert but it is not going to be installed
 gpac : Depends: gpac-modules-base (= 0.5.2-426-gc5ad4e4+dfsg5-5) but it is not going to be installed
        Depends: libgpac4 (= 0.5.2-426-gc5ad4e4+dfsg5-5) but it is not going to be installed
 libapache2-mod-php7.3 : Depends: php7.3-common (= 7.3.4-2) but it is not going to be installed
                         Depends: php7.3-json but it is not going to be installed
                         Depends: php7.3-opcache but it is not going to be installed
                         Depends: apache2-api-20120211
                         Depends: apache2-bin (>= 2.4.16) but it is not going to be installed
 motion : Depends: libmariadb3 (>= 3.0.0) but it is not going to be installed
          Depends: libmicrohttpd12 (>= 0.9.50) but it is not going to be installed
 php7.3 : Depends: php7.3-common but it is not going to be installed
 php7.3-cli : Depends: php7.3-common (= 7.3.4-2) but it is not going to be installed
              Depends: php7.3-json but it is not going to be installed
              Depends: php7.3-opcache but it is not going to be installed
              Depends: php7.3-readline but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

I then tried to change server to apache2 and I get something about cant read motion no such file (too fast to copy) and then it asks me yes or no to start camera system now.  I select yes but it says php not found.  I'm connected to a guest wifi that doesn't require password if that means anything.  Please tell me what I'm doing wrong.  I tried to go to http://ipAddress:80/html and http://ipAddress but it says This site can't be reached, connection timed out.  Thanks for your help!

roberttidey, 2019-10-23:
Something went seriously wrong there!

Had you installed anything else on the Pi first? That might have loaded other stuff on first that is causing things to get very confused.

If so try re-imaging a clean buster raspbian on to the SD card and then follow wiki instructions.

If it still fails then let me know and I'll do some further checks.

7176462, 2019-10-24:
Hi again.  I just tried to comment on the github site but it didn't let me
post.  I'm not sure what happened on the first install but i had done that
before this latest one so maybe somethings have been fixed since then.

I finally finished imaging a clean buster onto a new SD card and followed
the wiki again. It now gives me this message when I leave the defaults on
the blue screen:
Building dependency tree
Reading state information... Done
zip is already the newest version (3.0-11).
zip set to manually installed.
The following additional packages will be installed:
  apache2-bin apache2-data apache2-utils gpac-modules-base libapr1
libaprutil1 libaprutil1-dbd-sqlite3 libaprutil1-ldap libfreenect0.5
  libgpac4 libmariadb3 mariadb-common mysql-common php-common php7.3-common
php7.3-json php7.3-opcache php7.3-readline ssl-cert
Suggested packages:
  apache2-doc apache2-suexec-pristine | apache2-suexec-custom php-pear
default-mysql-client postgresql-client openssl-blacklist
The following NEW packages will be installed:
  apache2 apache2-bin apache2-data apache2-utils gpac gpac-modules-base
gstreamer1.0-tools libapache2-mod-php7.3 libapr1 libaprutil1
  libaprutil1-dbd-sqlite3 libaprutil1-ldap libfreenect0.5 libgpac4
libmariadb3 mariadb-common motion mysql-common php-common php7.3
  php7.3-cli php7.3-common php7.3-json php7.3-opcache php7.3-readline
ssl-cert
0 upgraded, 26 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,771 kB/8,313 kB of archives.
After this operation, 28.1 MB of additional disk space will be used.
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf
apache2-bin armhf 2.4.38-3+deb10u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf
apache2-data all 2.4.38-3+deb10u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Err:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf
apache2-utils armhf 2.4.38-3+deb10u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Err:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf apache2
armhf 2.4.38-3+deb10u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Failed to fetch
http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2/apache2-bin_2.4.38-3+deb10u1_armhf.deb
 404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Failed to fetch
http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2/apache2-data_2.4.38-3+deb10u1_all.deb
 404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Failed to fetch
http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2/apache2-utils_2.4.38-3+deb10u1_armhf.deb
 404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Failed to fetch
http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2/apache2_2.4.38-3+deb10u1_armhf.deb
 404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Unable to fetch some archives, maybe run apt-get update or try with
--fix-missing?

When I change apache to apache2 on the blue screen, I get the can't read
motion.conf no such file or directory again. and then it asks for yes or no
to start the camera and bash: php: command not found

pishotta, 2019-10-24:
I'm just an observer,  but I've had problems in the past when the Raspbian installer tries to use IPV6 to contact the repos.  I forget how, but there is some way to force the system to use IPV4.  That might help.-- Fred P.

http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2/apache2-bin_2.4.38-3+deb10u1_armhf.deb

http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2/apache2-data_2.4.38-3+deb10u1_all.deb

http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2/apache2-utils_2.4.38-3+deb10u1_armhf.deb

http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2/apache2_2.4.38-3+deb10u1_armhf.deb

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/525?email_source=notifications\u0026email_token=ACROKDNQ6SVWXPK4BVKJ7E3QQIAPFA5CNFSM4JEKLEK2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECGKQ7I#issuecomment-546089085",

"url": "https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/525?email_source=notifications\u0026email_token=ACROKDNQ6SVWXPK4BVKJ7E3QQIAPFA5CNFSM4JEKLEK2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECGKQ7I#issuecomment-546089085",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

7176462, 2019-10-25:
Good news!!

Robert and Fred, Thanks for the suggestion.  I didn't try Fred's IPV4 suggestion but I did find this:
https://www.raspberrypi.org/blog/buster-the-new-version-of-raspbian/
 
Reading further down, Dan suggests trying this command:
sudo apt-get update --allow-releaseinfo-change

Thanks Dan!

It worked and updated all the armhf files after several minutes.  I think the errors were due changes in the apache folder:
http://raspbian.raspberrypi.org/raspbian/pool/main/a/apache2/
where there's only "apache2-data_2.4.38-3+deb10u3_all.deb" and not"... deb10u1..." listed.

Thanks again to everyone who helped!

closed 2019-10-25 by 7176462