silvanmelchior/RPi_Cam_Web_Interface #526: sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set

4920441, 2019-10-30:
Hi,

I just did a fresh install of debian/raspbian buster and rpi-cam-webinterface.
After installing thelatter and the reboot I got  a....:

sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set

Erm... yes, that should not be....

pi@raspberrypi:~ $ ls -alh /usr/bin/sudo
-rwxr-xr-x 1 www-data www-data 145K Oct 12 14:49 /usr/bin/sudo

I doubt debian nginx packet was the source of this, so maybe there is a bug in the install script somewhere?  sudo shuld surele notbelong to www-data at all....

Cheers,

roberttidey, 2019-10-30:
The install.sh does not change sudo rights or ownership on that file and on my installs that file is still owned by root.

It does add a file to sudoers.d to give www-data sudo rights to run date, shutdown and reboot.

I can only think that some dependency going on during the install is changing the ownership. Which web server are you selecting during install? If not Apache then maybe worth trying using that to see if that makes a difference. Note that when changing web server it is best to start from clean raspbian.

4920441, 2019-10-30:
Hi,
Don't know what is happening then.

I just copied a fresh raspbian lite image  on the pi (and added my public key to the root user, so the thing should be a shoot in the foot again....)  and tried again as pi user to install with the appropriate sudo rights.

I also installed as much as I could remember in advance but after install.sh had run, it was the same effect:

sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set
pi@raspberrypi:~/RPi_Cam_Web_Interface $ 

The first approach was by using the root user for installing, but this time I totaly related on sudo for installing...

4920441, 2019-10-30:
Hi,
I just spotted something which should be normal....:

(stock raspbian buster lite installation so far, just rebooted for the first time and this time I chose apache for installation...:)

raspimjpeg: no process found
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package php5 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package php5-cli is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libapache2-mod-php5 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libav-tools is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  ffmpeg

E: Package 'php5' has no installation candidate
E: Package 'php5-cli' has no installation candidate
E: Package 'libapache2-mod-php5' has no installation candidate
E: Package 'libav-tools' has no installation candidate
sudo: a2enmod: command not found
^C
root@raspberrypi:~/RPi_Cam_Web_Interface# ^C

I am wondering if I use the wrong git...  php5 should be obsolete since stretch.

roberttidey, 2019-10-30:
Yes. Use install.sh The one beginning with RPi was contributed by another user a long time ago and is out of date