silvanmelchior/RPi_Cam_Web_Interface #529: A lot of warning/errors at /var/log/messages etc at Raspberry Pi 2B+ with standart RPI camera

thc2018, 2019-11-21:
Good day! RPI-Web-cam interface version 6.6.6
Raspberry Pi standart camera connected to Rpi 2B+, everything works well!
But in /var/log files, such as syslog, messages and user.log a lot of warnings.
messages:
`Nov 21 16:19:27 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:19:27 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:19:27 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:19:27 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:19:28 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close'                                        )
Nov 21 16:19:28 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:19:28 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close'                                        )
Nov 21 16:19:28 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:19:28 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:19:28 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:19:28 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:19:28 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:19:28 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close'                                        )
Nov 21 16:19:28 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:19:28 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close'                                        )
Nov 21 16:19:28 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:19:28 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:19:28 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:19:28 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:19:28 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
`
in syslog:
`Nov 21 16:25:51 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:25:51 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Nov 21 16:25:51 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:25:51 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:25:51 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:25:51 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:25:51 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:25:52 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Nov 21 16:25:52 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:25:52 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Nov 21 16:25:52 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:25:52 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:25:52 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:25:52 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:25:52 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:25:52 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Nov 21 16:25:52 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:25:52 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Nov 21 16:25:52 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:25:52 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
`
in user.log:
`Nov 21 16:26:25 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:26:25 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:26:25 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Nov 21 16:26:25 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:26:25 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:26:25 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:26:25 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Nov 21 16:26:25 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:26:25 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Nov 21 16:26:25 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:26:25 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:26:25 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:26:25 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:26:26 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
Nov 21 16:26:26 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Nov 21 16:26:26 raspberrypi motion: [1:nc2] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:26:26 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
Nov 21 16:26:26 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
Nov 21 16:26:26 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
Nov 21 16:26:26 raspberrypi motion: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
`
Why? How can I remove this warnings/errors?

roberttidey, 2019-11-21:
Those look like messages being generated by the motion process which is only used if you select external motion detection.

Try it with internal motion detection which is the recommended method as it is much more efficient.

If you didn't have external motion selected then check the running processes with cmd ps -A

If motion is in the list then something else is starting it up which should not be happening.

thc2018, 2019-11-21:
I am using internal motion detection.
ps -A | grep motion

gives me a 2 motion processes:

>  684 ?        00:00:42 motion
>  691 ?        00:00:42 motion

thc2018, 2019-11-21:
I have killed process 691. Now, I have 1 motion process. Every thing works fine, but warning and errors are still incoming into log files

roberttidey, 2019-11-21:
In internal motion detection mode the software makes no use of the motion process and doesn't start it up. All the motion detection is done internally in the raspimjpeg process.

Did you install anything else that might have started up the motion process.?

thc2018, 2019-11-21:
No, I did not install anything, that can start motion. Only RPI-Web-Cam.
I have killed both motion processes. Every thing fine! No more warning and errors.

I have found an entry in rc-local:
#autostart motion
motion start

pishotta, 2019-11-21:
Check to see that the external motion service is disabled at the system level

 

                sudo systemctl status motion

 

It should say it’s disabled.  The fact that you are seeing the processes suggests that they may be starting at boot.

thc2018, 2019-11-21:
sudo systemctl status motion

> ● motion.service - LSB: Start Motion detection
>    Loaded: loaded (/etc/init.d/motion; generated; vendor preset: enabled)
>    Active: active (running) since Thu 2019-11-21 23:03:51 MSK; 6min ago
>      Docs: man:systemd-sysv-generator(8)
>   Process: 241 ExecStart=/etc/init.d/motion start (code=exited, status=0/SUCCESS)
>    CGroup: /system.slice/motion.service
>            └─703 /usr/bin/motion
> 
> ноя 21 23:09:56 raspberrypi motion[703]: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
> ноя 21 23:09:56 raspberrypi motion[703]: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
> ноя 21 23:09:57 raspberrypi motion[703]: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
> ноя 21 23:09:57 raspberrypi motion[703]: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
> ноя 21 23:09:57 raspberrypi motion[703]: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
> ноя 21 23:09:57 raspberrypi motion[703]: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.
> ноя 21 23:09:57 raspberrypi motion[703]: [1:nc3] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
> ноя 21 23:09:57 raspberrypi motion[703]: [1:nc3] [NTC] [NET] netcam_read_first_header: Non-streaming camera (keep-alive not set)
> ноя 21 23:09:57 raspberrypi motion[703]: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: disconnecting netcam since keep-alive not set.
> ноя 21 23:09:57 raspberrypi motion[703]: [1:nc3] [NTC] [NET] netcam_read_html_jpeg: leaving netcam connected.

pishotta, 2019-11-21:
Looks like it’s trying to run. you can try 

 

                sudo systemctl disable motion

 

I would also comment out that “motion start” line in /etc/rc.local.  Then reboot and check status again.

thc2018, 2019-11-21:
Thanks to all, solved in my case.

thc2018, 2019-12-02:
I have found the problem.

Then you are doing a fresh install on raspberry pi, motion also installs on Pi. And during this installation, motion configure rc-local to autostart then Pi is starting (motion enabled)

roberttidey, 2019-12-02:
I'll check that. Raspbian never used to install motion and start it up automatically.

It is possible they have added this but it sounds a bad thing to do.