silvanmelchior/RPi_Cam_Web_Interface #531: How to get rid of ?? with %a and user_annotate.txt

thymjan, 2019-12-05:
Can anyone give me a hint how to get rid of the two questions marks when using the "%a" option to implement text from user_annotate.txt in the annotations?

[image]

This is my configuration of the annotation line:
`%a * birdhouse %D.%M.%Y %h:%m:%s:%u %v * `

I even don't want the Line Feed. I think control characters like EOF/CR/LF are shown up here.

pishotta, 2019-12-05:
Maybe you should be sure the user_annotate.txt file doesn’t contain a LF and/or a CR at the end? Check it with a text editor.

 

 <https://user-images.githubusercontent.com/12277755/70261657-311f7d00-1793-11ea-93b1-d0cdc5ac9d33.png> Bildschirmfoto 2019-12-05 um 19 12 29

thymjan, 2019-12-05:
> Maybe you should be sure the user_annotate.txt file doesn’t contain a LF and/or a CR at the end? Check it with a text editor.

Thanks for your answer. I've checked my text-file with nano. There was a LF.
I'm writing the temperature value with echo in user_annotate.txt.
I found an option to suppress the LF:

`/bin/echo -n ' *' $t1 'C * '  > /home/pi/RPi_Cam_Web_Interface/user_annotate.txt`

[image]

With the "-n" option the LF disappeared but the two question marks are still there.

roberttidey, 2019-12-05:
? is used when a character is included in the annotation string that is not supported by the font rendering used by the MMAL software to generate the overlay pixels..

Can you zip and upload a copy of the user_annotate.txt file so hat it can be checked? Do not just copy and paste its contents because this could lose some of the character data.

thymjan, 2019-12-05:
So, here it is:
[user_annotate.txt](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/files/3929222/user_annotate.txt)

roberttidey, 2019-12-05:
In Hex the content of that file is 20 2A 20 54 3A 20 2D 30 2E 39 33 38 20 43 20 2A 20 C2 A0

Note the last 2 characters are above the normal 128 character ASCII set and this is what generates the ?s

The C2 character renders as Â A with circumflex. The A0 character is a non breaking space,

thymjan, 2019-12-05:
I do no explicit fill these special characters in. Where do they come from?
I just use the echo command. Is there a tool you know which may perform better?
Would it be possible to filter them out?

thymjan, 2019-12-05:
I found a way with printf:
`/usr/bin/printf ' %s' $t1 > /home/pi/RPi_Cam_Web_Interface/user_annotate.txt`

Annotation text:
` *%a C * birdhouse %D.%M.%Y %h:%m:%s:%u %v * `

Result:
[image]

roberttidey, 2019-12-06:
Good to hear you have resolved it using printf. 

The strange result from echo -n seems to be triggered by the trailing space. I think echo must be encoding that as a non breaking space character which then leads to the unsupported character. If I use 
/bin/echo -n ' *' $t1 'C *'

then I don't get the extra characters.

I also note you are putting the user_annotate.txt in /home/pi/RPi_Cam_Web_Interface/

The default location as set in /etc/raspimjpeg is to put it in /dev/shm/mjpeg/

The reason for doing that is that /dev/shm is a temporary RAM based filing system and means that it avoids extra Flash memory writes on the SD card particularly in the case where the user_annotate.txt is being frequently updated.

thymjan, 2019-12-06:
Thanks for the hint with the RAM based filing system. Changed it back to defaults. First I had difficulties with user rights. Now it seems to work after reboot.