silvanmelchior/RPi_Cam_Web_Interface #532: Boxing Queue problem: How to reset the Boxing Queue

thymjan, 2019-12-07:
I've did a reboot while boxing queue wasn't empty.

After reboot videos were added till boxing queue was full. Seems the boxing does not work properly any more. All new videos are marked as busy in the web interface.

How does a reset of the boxing queue work?

roberttidey, 2019-12-07:
Video files are recorded to the media folder as .h264 files. The boxing process converts them to .mp4 using a queue of commands. The queue is held in memory and is not persistent.

If he raspimjpeg process is stopped (e.g. by a reboot) when the queue is not empty then there will be files left over in the media folder with a .h264 extension which are not converted.

When raspimjpeg starts up it does a check for any .h264 files in the media folder and tries to create boxing commands for them and add them to the start of its boxing queue.

It sounds like something has gone wrong with the restart maybe caused by a corrupt .h264 file when the system was first rebooted and this is causing the boxing to stick.

I would move any .h264 files out of the media folder and then restart.

The moved .h264 files could be manually converted to .mp4 files if they contain anything valuable. That would also allow you to see which was causing a problem.

thymjan, 2019-12-07:
I have to reboot several times to get it back working. I think there were several corrupted *.h264 files. I deleted them one by another.
Some files show up with to long durations.
[image]

thymjan, 2019-12-07:
Doesn't the box queue recognise corrupted h264 files?
I can't find hints in the /var/www/html/scheduleLog.txt file.

By the way, shows the boxing queue up as raspimjpeg process in htop?
How can I recognize a boxing process is still working?

roberttidey, 2019-12-07:
The box queue is just a list of the h264 files to be processed. For the top item it launches the MP4Box executable as an asynchronous process followed by a rm command to remove the h264 file. It then monitors to see when the h264 file is deleted which signals that the overall process has completed. It is then free to launch the next one from the queue.

It should be possible to see the MP4Box process in the ps list and in top while it is running.

There is no easy way to check for corrupted files.

It might be possible to add a timeout to the process but some people do process quite large files so the MP4Box process can go on for a while. I also haven't had many reports on corrupted files during normal operation. I have 4 active cameras running for years and not had one.

Note that the process launched is actually configurable see MP4Box_cmd in /etc/raspimjpeg so it would be possible to do other stuff there.