silvanmelchior/RPi_Cam_Web_Interface #536: Deleting all the images

KoushikRaghav, 2020-01-08:
Hi, 

I need to delete all the images and keep the latest taken image. Pls guide me to write in php

roberttidey, 2020-01-08:
Let me see if I can clarify what you want to do.

Do you want to have an extra button on the preview page that deletes all but the last image?

Do you want to change the behaviour of the Delete All button to do that?

Do you mean image as in the last still image capture or could it also be a video?

For all of these you would need to modify the preview.php file.

Currently the delete All action is performed by the maintainFolders function in preview.php. You would need to modify that so that it could retain the last files you wanted. First you would need the most recent file thumbnail name. Then you could add a test for this in the maintainFolders code so that it was not deleted.

KoushikRaghav, 2020-01-09:
In cmd_pipe.php file, I have written code to delete all the old images and store the latest image only. I have interfaced Rpi Cam Web Interface into my application. So, that latest image(lastCapturedImage) is not returning to my application.
[image]

I request you to look into the code and guide me to delete all the images and returning the latest image to my application

roberttidey, 2020-01-09:
I don't know how you are integrating this but putting it in cmd_pipe.php seems a bit strange. It means the code is run whenever any pipe command is sent to the server.

However, the code seems to be deleting all the images before it is trying to return the last captured one. I think you should put the code to find the lastCaptured one first and then check and exclude it when doing the delete code. Note that an image actually consists of a thumbnail and its high resolution actual capture file so you need to think about what you want to return.

closed 2020-01-19 by KoushikRaghav