silvanmelchior/RPi_Cam_Web_Interface #537: Need Clarification in saving image

KoushikRaghav, 2020-01-09:
Hello,

I have to get the latest captured image but, I'm getting the previously captured image. How to get the latest captured image and how the image is saving in the MEDIA_PATH???

roberttidey, 2020-01-09:
When a still image or video is captured then 2 files are saved in the media path.  These are the actual captured file (mp4 or jpg) and an associated lower resolution thumbnail. They have the same names but the thumbnail has an extra extension like .i0000.th.jpg added on. The i indicates it is an image, the number is the index number and th.jpg identifies it as a thumbnail. One can get the associated capture file from a thumbnail name just be stripping off this part.

So one strategy is just to find the latest thumbnail by its timestamp and then you have the name of the file and can get its base capture filename.

To get the latest filename depends on what environment you are programming in (shell, php etc) A search for find latest filename will show various methods.

KoushikRaghav, 2020-01-09:
I had given a delay of 1 second and the latest captured image is returned successfully after the delay. The captured image is taking 1 second to get stored in the media folder. 

Is there any event that I can get the 'latest' captured image without giving the delay? If so it will be easy to get the latest captured image!!

roberttidey, 2020-01-09:
When an image is captured then it can run a job macro command when it starts to take an image (start_img,sh) and also when it is completed (end_img,sh). These job macros get passed the name of the image as a parameter.

The macro can then do anything it likes with this name, e.g. it can copy it somewhere else. 

Check the wiki under job macros for how to use these.

KoushikRaghav, 2020-01-10:
Thanks! Will check that

I also have thought of getting the latest captured image with respect to the time when it is stored in the media folder. I have looked for filemname and filecname functions that return the timestamp of the 'updated' image file. But in my case, I'm not at all going to update/edit the image. So can you please suggest me the PHP function that gets the 'timestamp' of the latest stored file in that folder so that I will remove the delay function and replace it with the generic function. Then, I will return that image file to assume that it is the latest captured and latest stored image. Thanks in advance

roberttidey, 2020-01-10:
The macro is probably the cleanest way of doing this, but if you want to do it in php then check out the first answer at https://stackoverflow.com/questions/10989362/reading-the-latest-created-file-php for a short clean approach.

The glob function returns an array of files from a wildcard search. It then adds on the create time stamp and sorts the array based on this so that the latest can be easily indexed. You would need to put MEDIA_PATH in and you could include .th.jpg on the end of the wild card if you want to get the latest thumbnail.

KoushikRaghav, 2020-01-16:
> When an image is captured then it can run a job macro command when it starts to take an image (start_img,sh) and also when it is completed (end_img,sh). These job macros get passed the name of the [...]

Where can I find start_img.sh and stop_img.sh files?

KoushikRaghav, 2020-01-16:
Thanks a lot

closed 2020-01-19 by KoushikRaghav