silvanmelchior/RPi_Cam_Web_Interface #539: Full Screen mode shows webpage background

jarrah31, 2020-01-25:
Hi Robert,

If I manually trigger full screen by clicking the live video image, the web page background either side of the video is plain white.

However, if I use the `fullscreen true` setting in /etc/raspimjpeg, I can see the horizontal menu bars and black title bar in the background.
e.g.
[image]

Do you know if it's possible to fix this by any chance please?
TIA

roberttidey, 2020-01-25:
Yes. The fullscreen config does not do the same thing as the client click. This is because the config controls what the webserver delivers and the client click does some actions via javascript in the client itself.

I will have a look to see if the server side can be made similar.

roberttidey, 2020-01-25:
I forgot there was a change put in some time ago that allows the start up to be varied according to the url used.

If you use raspiIp/index.php#full then it will start up in full screen mode just like if you clicked at the client.

jarrah31, 2020-01-26:
Works a treat, thank you very much!

closed 2020-01-26 by jarrah31