silvanmelchior/RPi_Cam_Web_Interface #546: Privacy mask

icooper9000, 2020-02-11:
Is it possible to add privacy mask (as configurable in Motion) to the GUI.  Black out function would be sufficient, although an option to blur  the region would be useful too.

roberttidey, 2020-02-12:
It would be quite tricky to apply this into the main camera pipeline as this is all done in the GPU. It might be possible to apply to the medium resolution preview data but doing it for the high resolution feed would be very CPU intensive.

A different approach that would be fairly straightforward would be to do it with mask-image styling in the html.

But you also have to consider what you want the masking applied to. If the GUI is locked to preview only then that is all you have to worry about, but if it allows looking at captures and recordings then it gets trickier if you want those masked as well.

icooper9000, 2020-02-13:
Thanks for the reply. Ideally it would mask recordings and live feed. 

 

The annotation banner (with time, date, etc) is in live feed and in recordings. How is the annotation banner added ? (I’m not an expert here) Could a privacy mask be something crude and simple as that, but with a definable shape?