silvanmelchior/RPi_Cam_Web_Interface #549: Doesn't care about Existing Settings

Addicted5, 2020-02-28:
Hi, This install.sh replaces settings of web server (for me nginx) instead of appending necessary changes. Just removes default file from "/etc/nginx/sites-available" which was my server setting file & made me do extra work. There should be a manual method for people already having server setup.

mritou, 2020-04-10:
Hello,
Same for me and my apache config.
It break all the extisting configuration...
I try to deploy manually, but not really human friendly :)
Can you desribe how to deploy the webui for the camera without using install.sh script ?
Thanks for your work !

roberttidey, 2020-04-10:
install.sh does quite a lot of things; copying files into different places, setting permissions, arranging boot start up. setting up webserver etc.

If it is just the apache side you are concerned about then maybe the best way is just to stop from installing or touching apache settings. That is  easy to do by setting the webserver  name to anything other than apache, nginx, or lighttpd  when the install.sh runs (e.g. set it to fred). This way the install.sh won't do any webserver installation and won't call fnApache which is where the Apache settings are modified. All the other install stuff will be done including copying the web files under /var/www/html or whichever sub folder you have set.

You then need to do any apache changes yourself manually. You can see what the install.sh would normally do by looking at fnApache. It is basically setting up VirtualHost, port, sites-available and password stuff if required.