silvanmelchior/RPi_Cam_Web_Interface #550: [Question] Is there a possibility to use with another module (rather than raspimjpeg)

sbkomurcu, 2020-04-03:
Hi, I would like to ask if it is theortically possible to modify the program to use with another camera module? (i.e. VEYE-MIPI-290/327)
The camera module us using the natural connector however based on another driver.

Thank you so much in advance!

roberttidey, 2020-04-03:
The software that interfaces with the camera is  raspimjpeg (source code https://github.com/roberttidey/userland   under host_applications/linux/apps/raspicam )

This uses the MMAL interface layer provided by userland and which is used by the standard raspberry applications like raspivid etc. and exploits the power of the GPU.

If your camera driver supports using MMAL then it could be fairly straightforward. If it does not then the job becomes much harder.

sbkomurcu, 2020-04-10:
Hello again!  I have been working on the raspimjpeg codes since for a week now. Unfortunately I was not able to modify the source code in the way that to use another camera as input. I would like to ask a few more questions on this matter, if you don't mind.
 - Where exactly the raspimjpeg connects/picks the camera source? I have tried to change the mmal_component_t variables but I have ended with an error "undefined reference to {function}"
Thank you for your response, I appreciated!
Source code of he camera module can be found in the following link : [https://github.com/veyeimaging/raspberrypi/tree/master/veye_raspcam/source](url)

roberttidey, 2020-04-13:
The raspimjpeg module creates a camera object in the routine start_all in RaspiMCam.c

status = mmal_component_create(MMAL_COMPONENT_DEFAULT_CAMERA, &camera);

The standard raspberry camera allows two to be connected to a compute module but only one for standard raspberry pi so there is some code to pick one if more than one is present.

It then uses the camera object to configure it and to set up various processing chains in MMAL supported by the GPU.

You might want to look also at userland\interface\mmal\util\mmal_default_components.h  where there are different definitions of the MMAL_COMPONENT_DEFAULT_CAMERA