silvanmelchior/RPi_Cam_Web_Interface #552: malloc(): corrupted top size

oz-, 2020-04-20:
Hello, 

I'm unable to record any image or video by clicking on the buttons.
When I restart from the command line I have this message when sending a command from the interface.
I'm on Raspbian Buster.

Thank you!

roberttidey, 2020-04-20:
I've never seen that one before. It generally means either a buffer overflow or bad pointer. COuld be caused by some particular config.
 I'll need some more info before I can usefully help and replicate. Or, if you install again on a clean OS, it will probably just work.

1. How was it installed? What webserver did you choose. In particular did you install any other software?

2. Log and config. (scheduleLog.txt, uconfig, /etc/raspiconfig, schedule.json)

oz-, 2020-04-20:
Hello and thank you for your answer.
I finally managed to make it work by running the remove script and reinstalling everything with the install one.

1. I installed it at he beginning on a clean Lite install. I didn't install anything else and chosed Apache as web server, "without folder"...

2. Didn't make any change to the config files. Does the new scheduleLog (with working installation now) is still interesting for you?

The only thing I did was changing the camera settings before any other tests... 

Thank you! It's brilliant!

roberttidey, 2020-04-20:
Good to hear it's working. Probably some change to the settings got in an unexpected value.

No need to send anything unless you get further issues.