silvanmelchior/RPi_Cam_Web_Interface #558: Servos pan/tilt not working with latest version on Buster

ngmacha, 2020-05-05:
Hello - I had the RPi_Cam_Web_Interface working since 2018, running on RPi Zero with Raspbian Stretch and skalad's  fork for Servoblaster from https://github.com/skalad/RPi_Cam_Web_Interface_ServoBlaster_pan_tilt. The pan/tilt mechanism is custom, with servos connected directly to GPIOs.
Now I've decided to upgrade to Buster, fully re-imaged the SD card with the latest 2020-02-13-raspbian-buster-lite.img Raspbian, and installed RPi_Cam_Web_Interface afresh from this repo. Camera part is working perfectly, however now I can no longer control the servos from the web interface! The instructions to use ServoBlaster under https://elinux.org/RPi-Cam-Web-Interface are too vague - nothing works just renaming servo_off to servo_on, as seems actually ServoBlaster bit itself is not present? I've installed Servoblaster from https://github.com/richardghirst/PiBits/tree/master/ServoBlaster and was able to control the servos directly using echo P1-7=100 > /dev/servoblaster. But the direction buttons on the web interface still don't work. Please help, what am I doing wrong? 
Thanks a lot in advance - Mac Ha

roberttidey, 2020-05-05:
I am not aware of any change that Buster would have made to this. I suspect it may be a difference in the way ServoBlaster has been installed.

The camera software should be sending servo commands from the browser to the server component pipan.php. This translates them and sends them on to either pipan or servoblaster depending on what has been configured. As you have set servo_on then it should use servoblaster. Make sure that there is no pipan_on file present as that takes precedence.

The servo_on file should contain the current servo set up after the first button has been used. Make sure that has a little piece of json data with positions and configuration. If not then the buttons are not firing off the correct commands.

The pipan.php sends commands like 0=100 or 1=150 for the x axis or the y axis. Check those work using your echo test.

If neither of this helps then we may need to debug further. If you turn on developer tools in your browser then you should see the requests to pipan.php when you press the servo buttons.

ngmacha, 2020-05-05:
Thank you, @roberttidey for a quick reply! I've fully reimaged the SD card again, then first installed the ServoBlaster, and only then installed RPi Cam from this repo, and it is now working! So seem the only difference with my previous attempt was ServoBlaster installed after RPi Cam already running... Not sure why is it affecting. Anyway, thank you again, this issue could now be closed, I think - except maybe the documentations at https://elinux.org/Talk:RPi-Cam-Web-Interface needs to be updated - how is that done?

roberttidey, 2020-05-05:
Good to hear its working now.

Anybody can edit the wiki on eLinux.org Just create an account (free) and then you can edit it.

I encourage people to do this to add and clarify the information there.

closed 2020-05-05 by ngmacha