silvanmelchior/RPi_Cam_Web_Interface #560: Install/update doesn't work on Stretch

ras07, 2020-05-05:
When installing or updating on Stretch, you get the following error:

`raspimjpeg: /lib/arm-linux-gnueabihf/libc.so.6: version 'GLIBC_2.28' not found (required by raspimjpeg)
`

Is it still practical to support Stretch?

(See also https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=63276&p=1655663#p1655394 )

roberttidey, 2020-05-05:
Stretch is supported but you have to use a different version of raspimjpeg as the libraries are different in stretch. This is covered in the wiki instructions.

The stretch compatible version of raspimjpeg is included in the install but called raspimjpeg-stretch and must be renamed to the standard raspimjpeg. Normally this should be done prior to running the install script. If the install has already been done then do the renaming and run the install script again.

There are a couple of features (not that important) which are only available on buster installs.

ras07, 2020-05-05:
Ah, missed that - thanks!

closed 2020-05-05 by ras07