silvanmelchior/RPi_Cam_Web_Interface #561: How make stream public but not config ?

pbenaim, 2020-05-05:
I would like see the stream through internet without password, for example with cam_pic_new.php,
But I don't want that anybody can change any settings.
How can do that ? 
Set the password at install or reinstall, is set for all access, stream include... :-(

Any help ? Thanks a lot.

roberttidey, 2020-05-05:
You can set different views including the one with no access to settings via the user access levels described in the wiki.

Normally this is controlled by the user logging on to get a particular type of access, so one could have a general user / password with just the minimum view.

If you have one login for full access then any unknown user gets minimum access.

DidierMasse, 2020-05-06:
pbenaim, 

I make web page with two acces : only view and view + setttings protect ! 
If you want it i can give you

pbenaim, 2020-05-06:
Yes Please :-)
Merci Didier

DidierMasse, 2020-05-06:
give me some contact facebook/discord/mail, how you want and i send it
And after you can close this post (:

Tu es francais ?

roberttidey, 2020-05-06:
If you think this is generally useful then please add a link to the change to the wiki or if it is an alternative index page then I could maybe add it into the distribution so that people could use it more easily.

DidierMasse, 2020-05-06:
It's juste alternative page index, is not really secure i haven't experince for do this ! you can see here : charlesrpi.ddns.net

pbenaim, 2020-05-06:
> give me some contact facebook/discord/mail, how you want and i send it
> And after you can close this post (:
> 
> Tu es francais ?

Oui, Français 👌
But I don't want share here one way to contact me. Spam care 😒
Any other option?

DidierMasse, 2020-05-06:
sur discord c'est AngryDidierMasse#6899

DidierMasse, 2020-05-06:
@roberttidey Go see to charlesrpi.ddns.net:8000 and say me if you want it ( with theme personalisation :p )