silvanmelchior/RPi_Cam_Web_Interface #562: autostart

DidierMasse, 2020-05-06:
II have problem i think, RPiCam don't start after reboot, i have to start manualy, it's problem for me... 

Normaly it's automatic start no ?

roberttidey, 2020-05-06:
Yes. Autostart after reboot is an option when you install.

It places some start up commands into /etc/rc.local Check this file to see if it has the start up commands at the end.

DidierMasse, 2020-05-06:
i have this line : 

#START RASPIMJPEG SECTION
mkdir -p /dev/shm/mjpeg
chown www-data:www-data /dev/shm/mjpeg
chmod 777 /dev/shm/mjpeg
sleep 4;su -c 'raspimjpeg > /dev/null 2>&1 &' www-data
if [ -e /etc/debian_version ]; then
  sleep 4;su -c 'php /var/www/html/RPiCam/schedule.php > /dev/null 2>&1 &' www-data
else
  sleep 4;su -s '/bin/bash' -c 'php /var/www/html/RPiCam/schedule.php > /dev/null 2>&1 &' www-data
fi

Program give 777 ?!

roberttidey, 2020-05-07:
the rc.local looks fine.

Either is is not being executed at start up or the commands are not working for some reason.

Have you changed anything in standard set of raspbian?

You can also try putting in some dummy command like another mkdir in home directory to see if it is getting executed at boot time.

BETLOG, 2020-12-27:
nano /etc/raspimjpeg
# motion_detection false
motion_detection true