silvanmelchior/RPi_Cam_Web_Interface #563: Time-lapse video conversion is not working

jvlobo, 2020-05-07:
Hello there :)

I'm trying to get a video time-lapse but when I click on *Start convert* the new thumbnail gets stuck on *Busy*.

I've checked *scheduleLog.txt* and I can see the command that executed for the conversion

```
[2020/05/07 16:00:22] start lapse convert:(/usr/bin/ffmpeg -f image2 -i /var/www/html/media/t0000/i_%05d.jpg /var/www/html/media/tl_0000_0001_20200507_150439.mp4 ; rm -rf /var/www/html/media/t0000;) >/dev/null 2>&1 &
[2020/05/07 16:00:22] Convert finished
```

But it says finished at the same time it gets triggered, which is weird. I copied the command to run it myself: `/usr/bin/ffmpeg -f image2 -i /var/www/html/media/t0000/i_%05d.jpg /var/www/html/media/tl_0000_0001_20200507_150439.mp4`:

[image]

And that is true, there is no directory `t0000` inside `media`: 

[image]

Any idea of what is going on here? 
This is a fresh installation of `RPi_Cam_Web_Interface`, am I missing any extra configuration?

Thanks!

roberttidey, 2020-05-07:
I did a quick test and it seemed to work for me.

The actual time lapse images are normally in the main media folder. When a convert is started then a subfolder is created (like t0000) and the files copied (symlinked) into there and renamed to i_xxxxx.jpg. AFter the conversion is done then the folder and all these temporary files are removed. So the separate manual run of the command is not going to work because those temporary files will no longer be there.

One thing you can do is to temporarily change the preview.php code so that this removal is not done. This will allow you to check out the command manually.

To do this edit the preview.php in /var/www/html

Around line 237 there should be a line

$cmd = "(" . str_replace("i_%05d", "$tmp/i_%05d", $cmd) . " " . BASE_DIR . '/' . MEDIA_PATH . "/$vFile ; rm -rf $tmp;) >/dev/null 2>&1 &";

Remove the rm part like

 $cmd = "(" . str_replace("i_%05d", "$tmp/i_%05d", $cmd) . " " . BASE_DIR . '/' . MEDIA_PATH . "/$vFile ) >/dev/null 2>&1 &";

transfinite8, 2020-05-08:
I'm getting the same issue. I click on Start convert,  it creates a 48 byte mp4 file every time.

I was successfully creating time lapse files the last time I used the software (in September). I did an apt upgrade and ran update.sh yesterday. That combination seems to have broken something. Everything else in RPi_Cam_Web_Interface seems to be working correctly.

jvlobo, 2020-05-08:
Thanks for your reply @roberttidey 
So I followed your advice and when trying to convert it manually:

[image]

Seems like a problem with permissions which, on the terminal, make sense I guess, so I run it again with the user `www-data` and that seemed to work, I'm able to watch the video on the web interface:

[image]

Here you can see the owner and permissions for each directory:

[image]

HaukeRa, 2020-12-31:
I face the same issue. The log file says it's finished converting in the same second it started.

Though ffmpeg is stuck busy and never seems to finish. I tested this on a small 22 frame timelapse to ensure there are no resource constraints.

celynw, 2021-06-17:
I've still been running into this issue.
I tried messing with permissions, changing `preview.php`...
I think it just turned out that the default `convertCmd.txt` isn't suitable for the low-end devices like the Raspberry Pi Zero.

The "Convert finished" message is printed to the log immediately, and the return value of `ffmpeg` isn't used anywhere.
My pi was running out of RAM (I think) and killing the process, even with very short timelapses.
Also, the "Busy" tile in the file view appears as soon as there's a thumbnail, named `*.mp4.*.jpg`, and persists, whether it's actually busy or not.

So by just limiting the output resolution with `-vf scale=512:-1` it completes properly. Although it still says "Convert finished" immediately.

roberttidey, 2021-06-18:
The ffmpeg conversion is quite compute intensive which is why it struggles on lower spec pi.

Originally it used a gstreamer method which was much more efficient and faster as it used the GPU to do the heavy duty work. Unfortunately, some years back something got broken in gstreamer which caused errors with this and this is why it was switched to ffmpeg.

It is possible that gstreamer has been fixed so it might be worth given that another go.

The cmd template to use gstreamer was

gst-launch-1.0 -v multifilesrc location=i_%05d.jpg caps="image/jpeg,framerate=(fraction)30/1" ! jpegdec ! queue ! videoscale ! video/x-raw ! queue ! omxh264enc target-bitrate=15000000 control-rate=variable ! video/x-h264, profile=high ! h264parse ! mp4mux ! filesink location=

This needs to be put into the convertCmd.txt file in place of the ffmpeg version

gstreamer may need to be installed depending on the OS version