silvanmelchior/RPi_Cam_Web_Interface #564: nginx not starting

danito, 2020-05-15:
Hi,
when chosing nginx on a fresh install (v6.6.9), the file /etc/nginx/sites-availables/default is missing but there's a link to it in sites-enabled/ so nginx won't start. I removed sites-enables/default and nginx will start normally.
In install.sh I could see
```
sudo mv /etc/nginx/sites-available/*default* etc/nginx/sites-available/ >/dev/null 2>&1
```
which not seem to work as expected.
Thanks

Daniel

roberttidey, 2020-05-18:
This used to work so I guess something has changed on the nginx. I normally use Apache so I'll have to do another test install of nginx on latest version.

roberttidey, 2020-05-18:
I just did a quick test and after the nginx install I did have a /etc/nginx/sites-availables/default file together with the link from sites-enabled.

Had you done anything else prior to running the install.sh?

rnsi, 2020-05-21:
Same problem here. I've installed an OLED display and a UPS-Lite on a Pi Zero W. nginx doesn't start. Apache did start, but I reflashed and started over with nginx for performance.

rnsi, 2020-05-22:
I re-flashed a Zero W with 2019-09-26-raspbian-buster-lite, enabled the camera, and followed the update and install instructions. Nginx does not start. So this is not due to any changes I made, it's a stock install.

Any advise?

roberttidey, 2020-05-22:
This is under investigation. Advice at the moment is to use Apache until it is checked out.

roberttidey, 2020-05-22:
This was an issue with sites-enabled in nginx. default is removed from sites-enabled. Leaving the link in sites-enabled seemed OK with previous versions of nginx but this stops it starting with current version.

The install script now removes the link file in sites-enabled and nginx now starts OK from a fresh install.

If you want to resolve it manually with an existing install then do
sudo rm /etc/nginx/sites-enabled/*default*

and then start nginx with 
sudo service nginx start

rnsi, 2020-05-23:
That worked. Reinstalled from a fresh image.  However, when I move the camera, I get:
"Error in RaspiMJPEG: 
Restart RaspiMJPEG (./RPi_Cam_Web_Interface_Installer.sh start) or the whole RPi."

Seems like it doesn't liked to be moved. Or Motion is crashing it. Possible that a Pi Zero W just doesn't have the performance necessary to run this?

roberttidey, 2020-05-23:
Nearly all my cameras run on PiZero and they work fine. A lot of the work is done on the GPU and the CPU usage is actually very light (~5%).

I recommend using the internal motion detection method which uses much less resources than the external setting.

With crashes it is always worth checking 2 things. First are the camera ribbon cables secure including the little ribbon on the camera itself. The connector on the Zero is quite fragile. Second it is important that the power is good which means not only a good 5V supplybut also a good connection from the supply to the zero. If you are using a USB cable many of these have significant resistance and cause substantial voltage drop. Make sure you use a good quality short cable.

rnsi, 2020-05-24:
Looks like re-seating the camera cable solved it. I wouldn't have suspected that because the camera was working.

Thanks for an interesting project.