silvanmelchior/RPi_Cam_Web_Interface #565: Changing media path to remote drive

Bromley47, 2020-06-01:
I have a WD Drive attached to a pc - I have mounted the drive and can see it when I ls. But cannot find a way of saving media files to the hard drive and not the SD card.

Any help appreciated.

roberttidey, 2020-06-01:
Easiest way is to link the folder on the WD drive to the media folder. That way the rest of the system configuration remains unchanged but the media actually resides on the hard drive. This can be done in fstab.

Look in wiki under faq for specific advice on sharing. In particular, look at the boxing_path facility. This allows the live recordings to be made to SD and then to be moved to the final destination when they are boxed. This avoids any difficulties that may be caused by any temporary network bandwidth problems. The live recording part does need to have sustained high speed access.

Bromley47, 2020-06-02:
Thanks for the reply Robert - I assume you mean the wiki for the RPi interface? I can see the part about changing the path etc but I am unsure how to link the WD drive folder to the media folder..! I can edit fstab ok and I see media_path but what then?

Any help again appreciated (Nooby here!) and do i have to do in the 'root' or not?

Bromley47, 2020-06-02:
More info - this is what i have in fstab:

proc            /proc           proc    defaults          0       0
PARTUUID=9db0b9d9-01  /boot           vfat    defaults          0       2
PARTUUID=9db0b9d9-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

//192.168.1.***/h /home/pi/h/picamera cifs username=MEDIA-PC\****,password=*********,workgroup=WORKGROUP,vers=1.0,users,auto,user_xattr 0 0

I can see media_path in /etc/raspimjpeg...

Bromley47, 2020-06-02:
Now changed fstab entry to:-

//192.168.1.***/h/picamera /var/www/html/media  cifs username=MEDIA-PC\****,password=********,workgroup=WORKGROUP,vers=1.0,users,auto,user_xattr 0 0

but still not saving in picamera folder on WD Drive (h)..??

Bromley47, 2020-06-02:
ok Robert - have got the Download File page to show the volume of the WD  Drive but now when I click on record image no image is taken (in fact the record image button does not change colour) and if I click on the record video start the button  changes colour but no video is recorded...help - what have i done!!

Bromley47, 2020-06-02:
Going to reinstall and start again..watch this space!!

Bromley47, 2020-06-02:
Back to where I began still cannot save to USB hard drive - and of course cannot remember how I got to get the hard drive to show on the File page and then the recording didn't work - now the buttons work and the card is showing as the SD one in the Pi........frustrating or what..! Help needed - thank you

roberttidey, 2020-06-05:
I notice on your more recent issue you seem to have this working is that right?

An important point here is that ownership and permissions on any share / remote drive must give user www-data the ability to write. Sometimes this can best be achieved by ensuring any user can write.

Bromley47, 2020-06-06:
Yes Robert - I have got the camera to save images to the USB hard drive so that seems to be going OK- just the other problems with jerky video and speeded up play back to dirt now but thanks for your continued interest

BETLOG, 2021-01-10:
Maybe it will be helpful for you to see what I recently set up.
It's still not perfect, but it works to the point I haven't bothered to alter anything... yet.

Note:
 - 'history|grep chown' shows that I have made no changes to folder ownership on either machine, and yet www-data clearly has ownership in the right places.
 - Yes. It took a while to figure out. My attempts to force uid/gid ownership in the cams /etc/fstab were unecessary and actually stopped it from working. Less was better.
 
----
ssh pizerocam3

```
pi@pizerocam3:~ $ cat /etc/fstab
...
betlognuc:/media/user/video1/securityCam/new /var/www/html/media nfs defaults,nofail,x-systemd.device-timeout=9
```

```
pi@pizerocam3:~ $ ls -al /var/www/html/media
total 408
drwxrwxr-x 2 www-data www-data   4096 Jan  9 23:18 .
drwxr-xr-x 7 www-data www-data   4096 Jan  9 12:32 ..
-rw-r--r-- 1 www-data www-data 391629 Jan  9 23:18 2021-01-09--23-18-49_pizerocam3.mp4
-rw-r--r-- 1 www-data www-data  11698 Jan  9 23:18 2021-01-09--23-18-49_pizerocam3.mp4.v0000.th.jpg
-rw------- 1 pi       pi          232 Jan  8 21:44 .directory
```

----
ssh betlognuc

```
user@betlognuc:~$ cat /etc/fstab
...
/media/user/video1/securityCam/new  pizerocam*.lan(secure,rw,sync,fsid=0,no_root_squash,no_subtree_check)
```

 - where '/media/user/video1' is a 4-bay USB3 enclosure plugged into betlognuc

```
user@betlognuc:~$ cat /etc/exports
/media/user/video1/securityCam/new  pizerocam*.lan(secure,rw,sync,fsid=0,no_root_squash,no_subtree_check)
# always reload exports after editing /etc/exports
# exportfs -vr 
```

```
user@betlognuc:~$ ls -al /media/user/video1/securityCam/new
total 440
drwxrwxr-x 2 www-data www-data   4096 Jan  9 23:18 .
drwxrwxr-x 3 user     user      36864 Jan  9 18:52 ..
-rw-r--r-- 1 www-data www-data 391629 Jan  9 23:18 2021-01-09--23-18-49_pizerocam3.mp4
-rw-r--r-- 1 www-data www-data  11698 Jan  9 23:18 2021-01-09--23-18-49_pizerocam3.mp4.v0000.th.jpg

```
 - -rw-r--r-- is slightly tedious if you want to edit any existing files in place, but i havent bothered to fix that as it's not a bad safeguard.

The main bulk of my archived video are in:
/media/user/video1/securityCam
and the subfolder:
/media/user/video1/securityCam/new
is where RPCWI writes to, i manually move files i want to keep down one dir...again, sort of tedious, but it protects the 'keepers' and allows me to use 'delete all' within the RPCWI interface on the daily/weekely/current subset of video.