silvanmelchior/RPi_Cam_Web_Interface #566: Preview video very 'jerky

Bromley47, 2020-06-05:
The preview video looks very jerky and recorded videos play back at probably double speed!

Also the web page buttons take ages to respond i.e before they change colour.

This didn't happen when I first installed - I was impressed with the smoothness of the preview video and the recorded one.

I have Raspberry Pi Zero W and the official Raspberry camera. Captured images are transferred to a WD Hard drive with USB to Windows 10 PC.

I play back with VLC Media player.

Any help appreciated.

roberttidey, 2020-06-05:
The main thing that will affect preview smoothness is the speed and reliability of the network connection between the raspberry pi and the client. The processing power of the Pi Zero is quite sufficient.

If you are using a wifi connection then (pi Zero W) then the wifi link needs to provide a fast smooth connection.

One can adjust the bandwidth used by the preview by adjusting the preview settings. The three items control the size of the preview, the quality and the frame rate. You can normally afford to lower the quality factor a bit which compresses the images more, The frame rate can be dropped a bit by changing the divider. E.g changing to 2 will halve the bandwidth while still being pretty responsive.

A good technique for checking on the preview performance is to turn on debug in the browser (F12). If you then use default streaming (under system settings) then you will see individual frame requests and how they are being handled by the network. Using the MJPEG streaming is good to turn off when not debugging as individual requests are not used and the preview arrives just as a stream.

Playback of recordings at a different speed is normally caused by having the recording and boxing video fps set to different frame rates.

Note that if you are recording direct to a network drive then you should use the boxing_path facility to avoid recording problems caused by not being able to stream high resolution video data straight over your network.

Bromley47, 2020-06-05:
Thank you for the reply Robert - I have changed the divider to 2 and the quality to 5. The size is still set at 512. 

I assumed these settings would  have no affect until I reboot.

However these setting are not stored when I reboot and they revert to 10 and 1.

Both recording and boxing are set at 25 fps.

 I am not sure how to access debug? If i press F12 I get no response- where is this setting?

Also not sure what to do with box_path facility. I can get into etc/raspimjpeg but not sure what I should do then?

At the moment boxing_path is set as:-

 **_'media_path /var/www/html/media'_**

The preview screen freezes at times and then jumps around a bit.

With these settings everything is very slow - If I click on any button it takes ages to react and change colour.

Are any of these problems due to the fact that I am saving images on the USB hard drive (Attached to the PC not the Raspberry)

I am pleased how far I have got with getting it to save to the hard drive instead of the SD card but not sure this might not be affecting things.  Bit of a nooby here so your input welcome.

Bromley47, 2020-06-05:
Out of interest the preview seems to stop/start in line with the time - every millisecond the frame view changes...??

Not sure if this helps

roberttidey, 2020-06-05:
Changes to preview settings take immediate effect and should persist.

The log (under schedule settings) should show the changes being made and these changes are stored in uconfig under /var/web/html.

Are you sure you have made the link in your external drive to the media folder and not to /var/www/html ?

That top level folder should remain on the SD card to make sure it can be accessed quickly

Bromley47, 2020-06-05:
Are you sure you have made the link in your external drive to the media folder and not to /var/www/html ?

That top level folder should remain on the SD card to make sure it can be accessed quickly

**Hello Robert - how do I check that the above is done?**

Bromley47, 2020-06-06:
These are the last few lines form the log :-

{2020/06/06 08:47:24} h264 buffers set to recommended 1
{2020/06/06 08:47:24} Set up internal detect width=82 height=61
{2020/06/06 08:54:56} Finished boxing /var/www/html/media/vi_0000_20200606_061237.mp4 from Box Queue at pos 13
{2020/06/06 08:54:56} Start boxing /var/www/html/media/vi_0001_20200606_064253.h264 to /var/www/html/media/vi_0001_20200606_064253.mp4 Queue pos 14
{2020/06/06 08:54:56} Removed item from Box Queue

Is that of help?

Bromley47, 2020-06-06:
Ref your comment re making the link to the media folder the following is from raspimjpeg:-

base_path /var/www/html
preview_path /dev/shm/mjpeg/cam.jpg
image_path /var/www/html/media/im_%i_%Y%M%D_%h%m%s.jpg
lapse_path /var/www/html/media/tl_%i_%t_%Y%M%D_%h%m%s.jpg
video_path /var/www/html/media/vi_%v_%Y%M%D_%h%m%s.mp4
status_file /dev/shm/mjpeg/status_mjpeg.txt
control_file /var/www/html/FIFO
media_path /var/www/html/media
macros_path /var/www/html/macros
user_annotate /dev/shm/mjpeg/user_annotate.txt
boxing_path
subdir_char @
count_format %04d

The media path shows both html and media and that is how it is shown in fstab:-

//192.168.1.***/h/picamera /var/www/html/media cifs defaults,rw,credentials=/home/pi/.smbcredentials,vers=3,uid=www-data,gid=www-data,iocharset=utf8 0 0

Apologies for bombarding you with 'stuff' ..!

roberttidey, 2020-06-06:
Link looks like it is just to media folder which is good.

For jerkiness have you tried changing preview settings to see the effect of lowering network usage.

If the settings don't take effect or don't persist then check the log and also look for the uconfig file in /var/www/html which should have any settings that have been changed.