silvanmelchior/RPi_Cam_Web_Interface #567: The "t" in ca 1 [t] has stopped working?

NovySan, 2020-06-05:
Hello, I've been using 'echo ca 1 59 > /var/www/html/FIFO &' to start capture and record for 59 seconds. Suddenly, and I'm not sure how, the length of time argument is being ignored. The capture still starts just fine but will record until I issue the ca 0 command.  Where should I start looking?

roberttidey, 2020-06-06:
Check in log where it should say something like

Capturing 59 seconds

NovySan, 2020-06-06:
[2020/06/06 15:03:23] Send ca 1 59
{2020/06/06 15:03:23} Capturing 59 seconds
{2020/06/06 15:03:23} Capturing with split of 3600 seconds
{2020/06/06 15:03:23} Capturing started
...
{2020/06/06 15:05:08} Capturing stopped

As you can see, I had to manually stop the capture with the Stop Button after 2 minutes when it should have stoppped by itself after 59 seconds.

It doesn't seem to make a difference if I enter the command in the scheduler or via echo on a command line. 

[20200606-150610-scheduleLog.txt](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/files/4740843/20200606-150610-scheduleLog.txt)

roberttidey, 2020-06-06:
The log says you also have split set up. That is designed to allow continuous recording but with a new video started every split period. It takes precedence over the one shot time limited recording.

If you set the split back to 0 (off) then the time limited capture should work OK

NovySan, 2020-06-06:
Ok, thanks. I thought the Split was just the limit the length of a video that was longer than the split. I didn't realize it would effect videos shorter than the split length.

I have eliminated the split and capture now stops properly. Thanks you. This might be a nice note to add in the documentation.

closed 2020-06-06 by NovySan