silvanmelchior/RPi_Cam_Web_Interface #569: missing information on RAW

chriskoups, 2020-06-16:
Hello,
I am interested in raw imaging using the web interface!

I found out that the information written on the raw section is wrong and is missing one field. I was able to import those manually but I prefer to have those included as soon as I take the pictures.

Wrong:
Camera Model name: **RP_**xx0000 instead of xx0000

Missing:
Make: RaspberryPi

I use this code to automate jpg to dng mapping.
https://github.com/schoolpost/PyDNG

roberttidey, 2020-06-16:
All this software does is to enable RAW mode via the camera MMAL control layer.

I am not sure if it is possible to set or change the metadata but I'll see if I can find out.

Might be worth posing the question more generally on the camera forum to see if metadata can be controlled.

chriskoups, 2020-06-16:
The metadata can be manipulated (as I am doing that manually by using exiftool)

What command is used for capturing the photos?
rapstill has the correct information, but raspiyuv is missing make...

roberttidey, 2020-06-16:
The images, videos etc are all handled by a process called raspimjpeg which uses the MMAL interface to perform all the sampling, manipulation and capturing processes using the same sort of functions as raspistill, raspivid. It is controlled by commands fed into a pipe.

I can see where in the code raspistill manipulates some metadata and am checking to see whether something similar can be incorporated.

chriskoups, 2020-06-18:
Also, is it possible to have a png format for saved pictures? it is lossless and can give better quality.

roberttidey, 2020-06-19:
png is problematic as the GPU does not support this. Theoretically it can be done by using the normal processor but it would be much slower.

Jpeg with quality turned up to 100 should be pretty close to lossless.