silvanmelchior/RPi_Cam_Web_Interface #571: Annotation Position and Font

juancarlocc, 2020-07-04:
Hi all,

Is there any way to change the position and font of the annotations?

Thanks in advance.

roberttidey, 2020-07-04:
The annotation is added by sending calls into the MMAL camera software and these do not provide for controlling the start position or the font.

You can control size, text color and background color.

As the core software centres the whole text then some very limited control of postioning can be achieved by putting spaces into the annotation string to shift to the right or putting spaces at the end to shift to the left.

juancarlocc, 2020-07-04:
@roberttidey Thanks a ton for your reply. I do hope that in the future these options will be added,

closed 2020-07-04 by juancarlocc