silvanmelchior/RPi_Cam_Web_Interface #572: Cannot convert .h264 to .mp4 with large file size over the network

juancarlocc, 2020-07-11:
Hi all,

I was trying to see what my cat was doing at night so I've recorded a whole evening for around 9 hours. The issue I'm facing is that when checked the next morning, I can only see a 20GB .h264 file. Also my network was really busy. So i checked `top` to see what my raspberry pi was doing and below was the top 2 things in the list:

`  786 www-data  20   0   12964   5404   4284 D  15.5   0.7   0:13.40 MP4Box`
`  136 root      20   0       0      0      0 I   9.9   0.0   0:04.85 kworker/u8:2-brcmf_wq/mmc1:0001:1`

**MP4Box** is was using 15.5% cpu and **kworker** was using 9.9%. Also I noticed that the ram usage was 26.5MB free of 1GB.
Please note that I'm using a network attached storage for the media folder so I'm not sure if the issue is somewhat connected to that. I've never had issue with shorter videos before. I think I've recorded 3 hours before and it worked just fine. 

Appreciate if someone can help me fix the issue.

Thanks in advance.

roberttidey, 2020-07-11:
I have never tried to handle a recording that long and I suspect it is the memory usage that is causing the problem and making the conversion very slow.

It may be best to move the h264 to a platform with more resources and convert it there.

The software has the facility (video split) to break up continuous recordings into smaller segments and I would recommend using this to keep the the individual recordings less than say 1 hour.

802compute, 2021-11-03:
I've encountered this same issue with a Timelapse recording (15 seconds per frame for more than four hours), and on a RPI Zero W I eventually overwhelmed the memory and swap, effectively crippling the entire RPi.  Only after manually pulling power was the RPi_Cam application usable again.

I suspect there is too little memory to effectively run the transcode from .h264 to .mp4 on anything less than a RPi 4b

roberttidey, 2021-11-03:
Originally the software used gstreamer to do the conversion which was very efficient as it used the GPU and would run fine on any Pi. Unfortunately an OS update stopped that working several years ago and the software switched to using ffmpeg where the conversion is done on the main CPU and becomes much more processor and memory intensive. A long time lapse will struggle on Pi's with slower CPU and less memory.

Personally I normally download the timelapse set onto a PC for conversion where ffmpeg runs quite fast. As I would normally do editing of the video anyway, this is no problem for me.

I may try to revisit the gstreamer method to see if the original problem has been fixed now as this did give good performance. The main software doesn't have to change as it is just a different convertCmd.txt to kick off the gstreamer conversion instead of ffmpeg