silvanmelchior/RPi_Cam_Web_Interface #575: Question: how to stop ongoing motion recording

Creat1veM1nd, 2020-08-13:
Hello,

I was not able to find an answer, maybe I overlooked it. 

I am setting up a motion detection. When set too sensitive, the recording started and is going on forever with no chance to interupt it. Is there a way to stop motion detection while recording ?

Thanks for any hint!

mteuscher, 2020-08-13:
Had a similar problem and solved it with the following GET request: ```
http://[CAM_IP]/cmd_pipe.php?cmd=md 0
```
Just replace [CAM_IP] with your ip. Probably works if you just enter it in your browser

roberttidey, 2020-08-15:
The GET method will work.

A more fundamental change could also work.

The enabling of buttons for the different states is controlled by script.js. In particular, ajax_status.onreadystatechange (around line 236) sets the button labels, actions and disable for each state.

In the case of active motion recording the state is md_video. The video and motion detect buttons are disabled here but could be enabled (set disable to false) instead to allow manual override.

Creat1veM1nd, 2020-08-17:
Thanks for the hints! I will try the GET hack next time.

Creat1veM1nd, 2021-01-08:
Sorry for late feedback! The hack with URL worked, thanks again.

closed 2021-01-08 by Creat1veM1nd