silvanmelchior/RPi_Cam_Web_Interface #576: When i remove it, it DELETES ALL my files!!!!!!!!!

Syize, 2020-08-17:
I just run:
sudo ./RPi_Cam_Web_Interface.sh
and I select remove
and I select hard mode
and IT DELETES ALL MY FILES in root path.
ALL!!!!!!!!!!!!!!!!!
Fine.

roberttidey, 2020-08-17:
RPi_Cam_Web_Interface.sh was an old method for install etc. The wiki says you should use ./install.sh and its associated scripts.

oleg31337, 2021-03-07:
Having your wiki totally unclear of how to install it makes people try the "old method"
You should either fix the "old method" or remove it from repository.

roberttidey, 2021-03-07:
I'm not sure I agree the wiki is "totally" unclear about the install method as the step by step Basic installation at the beginning of the wiki only refers to the supported ./install method. There is also a section on the old / original method which makes it clear that it should not be used

I do agree that the presence of the old method could confuse some users who do not really read the install instructions and there is also a section on start up behaviour which refers to the old script.

The reason why I have been reluctant to remove it is because it was a contribution by a third party who did quite a lot of work on it originally. It has not subsequently really be maintained but I left it there in case there was a desire to do so.

What I have done now for the moment is to rename the file so it is clearer it should not be used. I have also deleted the section in the wiki on start up behaviour which refers to it.