silvanmelchior/RPi_Cam_Web_Interface #579: Motion Mask not applied to Internal Motion Detection

emjas92, 2020-08-30:
First, I'd like to mention I've had plenty of experience with Linux, Raspberry Pi, and Motion, and I've done much troubleshooting to try solve this problem with the motion mask.

I'm running an older Rpi B with the latest DietPi and RPi Cam Web Interface 6.6.11, with a 1296x972 @10fps resolution and a pgm mask to block motion detection in multiple areas in view of the camera.

Although the schedule log shows the mask is loaded, it is not actually working and motion is still detected in the black mask areas.
I have even tried a completely black mask, but motion is still detected.
The Motion Vector Preview shows motion in the black mask areas too.

`{2020/08/30 20:17:22} Set up internal detect width=82 height=61
{2020/08/30 20:17:22} Motion mask /var/www/mask.pgm loaded
{2020/08/30 20:17:23} DEBUG 3`

The mask file content is 
`P5
82 61
255
`
Followed by a line of 5002 NUL characters (according to NP++)

My questions are:

- Is the mask format correct? (I've tried different pgm formats, P1-P6)
- Is the mask applied to Internal motion detection?
- Should the mask affect the Motion Vector Preview?
- Is the mask used in the raspimjpeg binary or somewhere else?
- What's the easiest way to see if the mask is working?

Thanks for your help.

roberttidey, 2020-08-30:
Mask format looks correct. Software expects first 2 bytes to be 50 35 (P5) then it searches for the string '255' then it loads width * height bytes into a mask buffer. If length of file is sufficient then it thinks mask is valid

Mask is only used by internal motion detection

Mask does not affect Motion Vector preview. That is just the raw vectors from the camera.

Internal motion detection including masking is carried out in raspimjpeg. More specifically it is in a module called RaspiMMotion.c in 

https://github.com/roberttidey/userland/tree/master/host_applications/linux/apps/raspicam

Last question is a bit trickier to answer as I haven't seen this problem before. One thing you can do is put %c %f in the annotation string as that gives a running display of motion changes and frames which do take into account the masking.

If you want to post zipped mask file then I can run a test on it.

I'll also run a test on latest version to check nothing has got broken.

roberttidey, 2020-08-30:
Tested with my own blank mask and that does work ok.

emjas92, 2020-08-30:
Thanks Robert for quick response and detailed explanations.

I have done some more experimenting and found that the mask works when I restart RpiCam with `start.sh`
The mask does not seem work when stopping/starting using the web interface, even though the schedule log says it's loaded.
I tested this with the full black mask and then my actual mask.

I have found that %c stays at '0005' when the movement stops. Should this return to 0?

Thanks again.

roberttidey, 2020-08-31:
I'll check that further.

What do you mean by stop starting via the interface? Stopping and starting motion detection or stopping and starting the camera system.

A low number like 5 for %c is quite normal, there can be some rounding errors and boundary conditions. I have it on my list to investigate that further. It does, however have little effect as the typical threshold is much higher.

emjas92, 2020-08-31:
I tried stopping and starting motion detection, the camera, and the schedule using the interface buttons, but the mask did not seem to be working.
I'll do some more testing myself too.

roberttidey, 2020-08-31:
That is strange. My test was:

Started up with motion detection off
maskimage file in place
Start motion detection from web interface

Log then showed mask loaded and it was functional.

emjas92, 2020-09-02:
I have done some more testing using the web buttons.

1. Camera running using `start.sh` with mask and motion detection enabled -> mask is working
2. Stop and start motion detection only -> mask is **sometimes not working**
3. Stop motion detection then stop camera, then start both -> mask is working

In some cases the mask appears to be working, but then later I'll find recordings of motion in the masked-off areas.

Another point to consider is I'm using the Sun-based schedule with `md 1` for each time period and only setting the `em` mode between 'auto' and 'night'.
Perhaps this schedule period change is breaking the mask functionality? I'll have to monitor it for a few more days.

emjas92, 2020-09-09:
I'm still experiencing an issue with the mask not working after the camera has been running for some time.

For example, I'll stop/start motion detection from the web interface and confirm the mask is working, then the next day I'll find recordings triggered in the masked-off area. Stopping and starting motion detection gets the mask working again.

This happens every few days (not every day).
I am no longer using the 'Sun based' schedule, but now use the 'All Day' schedule with `mz 1;md 1;em auto;`.
The Pi is not rebooting, and I'm not aware of any other software that could be interfering.

Any thoughts as to why the mask would stop working after some time?
The actual motion detection is working fine (unwanted recordings is better than no recordings).

roberttidey, 2020-09-09:
I have some cameras which switch motion on and off twice a day and the mask keeps going OK without any issue.

There is a pipe command which can be used to enable or disable the mask; mz 1 (disable), mz 0 (enable). That is not issued by the software itself but is to allow external control. It also puts a message in the log when used.

I'll look into putting some extra debug in to help track down what is happening in your case.

It would help if I can see you specific motion parameters. If you get the uconfig file from /var/www/html then that will have any changes from the default set up.

emjas92, 2020-09-10:
The mask usually stops working the next day which is why I'm thinking it has something to do with the schedule.

What's your experience with 24h motion detection?

Here's my uconfig

    annotation  %Y.%M.%D %h:%m:%s %c 
    anno_background 1
    rotation 180
    sensor_region_w 65450
    raw_layer 0
    quality 95
    divider 5
    video_width 1296
    video_height 972
    video_fps 10
    MP4Box 0
    MP4Box_fps 10
    image_width 1296
    image_height 972
    image_quality 95
    tl_interval 600
    image_path /var/www/rpicam/media/%Y%M%D_%h%m%s_%i.jpg
    lapse_path /var/www/rpicam/media/%i_%t_%Y%M%D_%h%m%s.jpg
    video_path /var/www/rpicam/media/%Y%M%D_%h%m%s_%i.mp4
    media_path /var/www/rpicam/media
    vector_preview 0
    motion_external 0
    motion_threshold 15
    motion_image /var/www/rpicam/mask.pgm
    motion_startframes 1
    motion_stopframes 100
    log_size 1000
    end_vid &end_vid.sh

In case you wondered, my camera has a column of white pixels at the edge hence the `sensor_region_w`, and I'm using my own script to do the MP4Boxing on another device.

emjas92, 2020-10-05:
Just a quick update to say that the mask has been working fine since I last posted.

This includes using the web buttons and FIFO commands to stop/start the motion detection and camera.
I made no config changes (that I remember) so I'm not sure what the cause may have been.

I'll close this ticket as there's no ongoing issue.
Thanks again for your assistance.

closed 2020-10-05 by emjas92