silvanmelchior/RPi_Cam_Web_Interface #580: No Thumbnails in "Download Videos and Images"

sorcierwax, 2020-09-02:
Hello, 

I've been using this software for quiet a while now and I've never add an issue with it, untill today (but I'm aware that I certainly am the issue :-) )
After a total crash of my NAS, I've changed it for a new one and have recreated the same share (the one my Raspberry Pi is mounting with fstab).
I can record videos and pictures, but in the Download section I've got no thumbnails at all, just the X icon like there is a broken link.
On my NAS the thumbnails are created and perfectly fine.

Maybe a permission issue ?

Thank you for your help

roberttidey, 2020-09-02:
The folder needs to be owned by www-data and have read write permissions.

Your fstab can include parameters like
rw,uid=www-group,gid=www-group

sorcierwax, 2020-09-02:
Tnahk You Robert for your quick answer.

I've double check, here is my fstab line : 

//192.168.0.15/cam /home/waxou/cam cifs defaults,rw,credentials=/home/waxou/.smbcredentials,uid=www-data,gid=www-data,iocharset=utf8 0 0

And /home/waxou/cam is owned by www-data with 777 permissions and still no Thumbnail, but the video and pictures are correctly recorded in the share though.

Weird, I think my NAS is haunted :)

roberttidey, 2020-09-02:
Strange.

Three things

If you do a ls /var/www/html/media/*.th.jpg
Do you see the thumnails listed? (If not installed under html then change the path)

If you browse to ip/media/im_0000_20200902_220625.jpg.i0000.th.jpg (substitute an existing thumbnail name

Have you changed any of the paths in the config file /etc/raspimjpeg? The best way is to leave these alone and do the share to map the NAS over these.

then do you see the thumbnail in the browser?

sorcierwax, 2020-09-03:
I can see the thumbnails in /var/www/html/media but I've got a 404 error when I try to browse one of them like http://ip:port/media/im_0000_20200902_220625.jpg.i0000.th.jpg and same 404 error when I try to list http://ip:port/media/. However I can list the thumbnails in http://ip:port/html/media but when I click on a link I've got a ERR_INVALID_HTTP_RESPONSE

As you mentionned, I had indeed changed the paths in the raspimjpeg (it was along time ago and I didn't remember, sorry).
I put back a default one, but still, same problems.

I think my install is quite f**cked up and I'd better install a clean one.

sorcierwax, 2020-09-05:
Ok so I dit a fresh install with the latest Raspian. 
I've got my network share mounted with fstab and owned by www-data : 

fstab line : //192.168.0.20/cam    /home/pouetos/Cam cifs rw,credentials=/home/pouetos/.smbcredentials,iocharset=utf8,uid=www-data,gid=www-data    0    0

ls -l returns :  drwxr-xr-x 2 www-data www-data    0 sept.  5 09:47 Cam

I did reinstall RPI Cam Web Interface with install.sh, setting up Apache on 8007 port and my credentials.
After that, I've run "./RPi_Cam_Web_Interface_Installer.sh update" to change the storage location.

And still the same issue, not a single thumbnails in the frontend but thumbnails and pictures are good everywhere else.

http://ip:port/html/media shows the files, but when I clic on them : ERR_INVALID_HTTP_RESPONSE  

But if I switch back to internal storage "/var/www/html//storage/media", everything is fine.

So if I store locally it works, If I store on a samba share it fails. Do you think it's a mounting issue ?

Thanks again for your help

sorcierwax, 2020-09-05:
Ok, back with this, I've also tried to mount directly the smb share to /var/www/html/media with no success either.
I've even tried to symlink, still the same.

roberttidey, 2020-09-05:
I wouldn't run ./RPi_Cam_Web_Interface_Installer.sh under any circumstances. This is an old script contributed by another party and hasn't been maintained (I should probably remove it) The last version did have some problems as well so I don't know if it could be damaging the installation in some way.

Having said that it does some like some sort of mount issue particularly as the browser won't access those files. What is puzzling me though is why it is specific to thumbnails. One possible thought is that maybe there is something about the filename length that is causing the issue when the mount is used. The thumbnail files have a slightly longer name. Ways of checking whether that would be the cause is to try either renaming a thumbnail file to something shorter and then see if the direct browser access or you could temporarily remove the %Y%M%D from say the image_path in /etc/raspimjpeg and then see if an image capture works.