silvanmelchior/RPi_Cam_Web_Interface #581: FPS is capped at 15

Skipher, 2020-09-09:
I have a Raspberry Pi 3B+ that can only record up to 1080p at 20 fps before losing frames. I really wanted to hit 30 fps, so I got a Pi 4. When I setup the RPI Cam Web Interface, the live preview was showing at 1080p at 15 fps. I increased the frame rates for recording and boxing to 30, but it is still streaming and recording only at 15 fps. If I play around with different numbers by gradually increase the frame rate, I can maybe hit 20 fps sometimes. I've tried decreasing the quality to even 5 or 1 to see if there was a performance limit, but that doesn't seem to be it. I don't know why the frame rate seems to be locked. I am still able to change other camera settings; it's just the frame rate that I can't affect.

roberttidey, 2020-09-09:
There is no specific cap on the frame rate in this software.

Are you saying that when you change the frame rate on the web interface then it comes back with a different number?

Note that the live stream (preview) is not the same as a video recording. It is a jpeg stream either of separate images or formed into a mjpeg stream. The latter is more efficient and is selected under system settings. Normally most users have the preview at less than full resolution (preview settings) and rely on the actual video recordings which take place at the full specified resolution. However one can up the preview width to match the recording width to get the higher resolution, This does increase the bandwidth needed for the preview stream and this can become an issue for some networks particularly if over wifi. One can see the effect of the preview by turning on developer tools in the browser to see the network activity. Here it is quite good to turn off the mjpeg stream temporarily as you will then see each frame being delivered over the network and can see the packet size.

Skipher, 2020-09-09:
Thank you for the response.

Here's what I'm experiencing. On the web interface, I will enter say '25' or '30' for the recording and boxing fps. When I click 'ok' to apply the setting, the stream reloads, but I do not see the stream play at a higher fps than around 15. I'm certain it's not a network bandwidth issue since recorded videos also have a similarly low fps of 15.

I've already used the same camera on my Pi 3B+, and it was definitely capable of streaming and recording at around 24 fps. I was expecting the Pi 4 to achieve a higher frame rate, given that it's a newer hardware and all. It's almost as if there's a configuration file or something that overrides my settings on the web interface.

closed 2020-09-13 by Skipher