silvanmelchior/RPi_Cam_Web_Interface #582: Raspberry performance monitor

fragarray, 2020-09-10:
Hi, i love your app.
One of my suggestion is to add a performance monitor somewhere. 
Some object (or a simple string) who can give us essential infos about the pi, like RAM Usage, CPU Temp and CPU Use (especially during timelaps rendering process who need a status bar too) like Pi-Hole web interface.

A place can be just inside the upper grey bar and have a result like:

RPi Cam Control v6.6.11: mycam@raspberrypi                                                                       Ram XX MB - Temp XX °C - Load XX%

Good work guys! ;)

BETLOG, 2020-12-27:
I'm sure theres been plenty of people do this, but if theres a contributions space for such things I don;t know of it.
I also *just now* added a similar thing.. so lets see if i can document it right:

`
piinfo.service
`

```
[Unit]
Description=RPiWebCamInterface user annotation service

[Service]
Type=oneshot
ExecStart=/bin/bash /home/pi/piinfo.sh
```

`
piinfo.timer
`

```
[Unit]
Description=RPiWebCamInterface user annotation timer

[Timer]
OnUnitActiveSec=1s
OnBootSec=1s

[Install]
WantedBy=timers.target
```

`
piinfo.sh
`

```
#!/bin/bash
# https://elinux.org/RPi-Cam-Web-Interface#Is_there_any_way_to_have_the_.22UPTIME.22_displayed_on_the_Web_Cam_Interface.3F
# for use with %a
t=$(( $(cat /sys/class/thermal/thermal_zone0/temp)/1000 ))
f=$(( $(cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq)/1000 ))
echo -ne "$HOSTNAME ${t}C ${f}MHz\\n " >/dev/shm/mjpeg/user_annotate.txt 
echo "$HOSTNAME ${t}C ${f}MHz" # so it appears on the cli if you execute it manually
```

```
# send from PC to pi
scp /home/user/documents/scripts/raspberry/etc/systemd/system/piinfo.* pizerocam1:/tmp
scp /home/user/documents/scripts/raspberry/home/pi/piinfo.sh pizerocam1:/home/pi
```

```
ssh pizerocam1

sudo su
mv /tmp/piinfo.* /lib/systemd/system/
ln -s /lib/systemd/system/piinfo.* /etc/systemd/system/
systemctl daemon-reload
systemctl start piinfo.timer
systemctl enable piinfo.timer

nano /var/www/html/uconfig
# add %a somewhere
annotation %Y-%M-%D--%h-%m-%s %a
exit # sudo su
exit # ssh pizerocam1
```

interface of the second pi i just cloned that includes this user %a annotation
[image]

roberttidey, 2020-12-27:
The elinux wiki can have material added by anyone. So this info is great to put into Additions and tricks section.

BETLOG, 2020-12-31:
https://elinux.org/RPi-Cam-Web-Interface#Implement_.27user_annotation.27_.28.25a.29_as_a_service