silvanmelchior/RPi_Cam_Web_Interface #583: My 30fps seems to be actually 25fps in a 30fps wrapper

RichShumaker, 2020-09-10:
I have searched in several places and was unable to pin down this issue.
i wanted to start a ticket before I do any more changes and experiments to my setup.
That way I can come back here and say what I did if I end up finding a way to make this work.

Currently when I record 1080p 30fps it seems to be recording 25p in a 30fps wrapper.
The footage seems sped up when I watch it and I have to time stretch it to match my off camera audio.

My H264 files are not being converted to MP4's so I am doing that in VLC on my PC.
When I watch the H264 files before conversion they seem sped up, same after conversion.

I am using a RasPi 4 1gb model with the HQ camera.
Just installed everything within the last month.
I have all my user setup pointed at 30fps, in my uPresets.html
My RPWCI HTML Page in the browser is set to 30 with 30 for my binning..

Thanks so much for everyone that has contributed to this software over the years it is simply amazing.

Skipher, 2020-09-11:
I also have this issue (#581 ). I had a Pi 3B+, and I noticed some frame drops at higher resolutions, which resulted in a 30 fps recorded video essentially being a lower fps video played at a faster speed. I also upgraded to a Pi 4 recently with fresh installs. However, my settings also seem to be not changing what fps it streams and records at. I've tried changing to fps settings in the raspimjpeg file in /etc and /var, but that didn't do anything. Where is the 'uPresets.html' located so that I may try that?

RichShumaker, 2020-09-11:
> I also have this issue (#581 ). I had a Pi 3B+, and I noticed some frame drops at higher resolutions, which resulted in a 30 fps recorded video essentially being a lower fps video played at a faster [...]

Thanks for the heads up.
When I hit the reset button in RPi Web Cam Interface it reverts back to the PAL 25 25.
I was looking for where that was set to see if that default setting could be causing some issue somehow.

RPi Web Cam Interface is only an overlay of what the Pi can do so I will try to do some testing to see if the underlying Capture has the same issue. I need to determine the best way to mirror my settings that I have set the camera for when I do test recordings commands directly so it is as close to apples to apples as possible.

RichShumaker, 2020-09-11:
I was able to do some testing.
I first updated  /etc/raspimjpeg
Updated to 30fps in this file as it is set to 25

I rebooted and tried to record again and the same results occurred.

Then I used this command on the command line after stopping my camera in RPi Cam Web Interface
raspivid -t 30000 -w 1920 -h 1080 -fps 30 -b 2200000 -p 0,0,1920,1080 -o pivideo.h264

This file seems to be recorded with no issues, meaning it plays back correctly it does not play back fast.

This leads me to believe that either I did something wrong in my RPi Cam Web Interface webpage(meaning I put a wrong number on a line like 0 for brightness or something less obvious as 0 for brightness returns a pitch black image)
Or that there is something that is overriding what we put in RPiCWI
Next up is 
Installation Tips
Step 6: Remove extra material from full raspbian install and update

sudo apt-get remove --purge wolfram-engine scratch minecraft-pi sonic-pi dillo gpicview oracle-java8-jdk openjdk-7-jre oracle-java7-jdk openjdk-8-jre
sudo apt-get clean
sudo apt-get autoremove
sudo apt-get update
sudo apt-get dist-upgrade

I rebooted after this and the issue persists.

RichShumaker, 2020-09-11:
More Testing
I defaulted everything to Auto.
I then tried 25fps with 25 binning. as well as 30fps and 30 binning.
Both play fast.
So frustrating as I just looked at footage I thought was good and nope.
I have had this issue for the entire time I have owned my HQ camera which is about a month or so.

RichShumaker, 2020-09-11:
I apologize for the piecemeal testing.
I streamed to my capture system to check to see if the stream had a hitch.
The recording was fine all though it was a much lower quality as I have my system set to 384 to reduce bandwidth.

Where or how do I test next, does anyone have a testing methodology so I can see where my issue might be?

roberttidey, 2020-09-12:
I think the first place to start is to do test recordings using raspivid. raspivid and raspimjpeg use the same methodology to access the camera and do recordings but raspimjpeg integrate in lots of other features like the preview stream, image recording, time lapse and motion detection.

If you see similar problems with raspivid then something is going wrong with the lower levels of the camera access.

It would also be worth trying to make raspimjpeg recordings with browser off

echo 'ca 1' > /var/www/html/FIFO   to start a recording and
echo 'ca 0' > /var/www/html/FIFO   to stop the recording

Skipher, 2020-09-12:
I've been able to successfully record 1080p 30fps videos through raspivid. It's only when I work with this program that I have the fps issue. I've confirmed that this is not a network issue, and I've tried on two different hardware: Pi 3B+ and Pi 4.

RichShumaker, 2020-09-12:
I too did a command line capture and the same issue occured. I will start and stop with the command you sent and not open the browser. Are there any other troubleshooting steps to try?
Thanks for your help.

roberttidey, 2020-09-12:
I just tried a 1080p 30fps recording boxed at 30fps. To eliminate any uncertaintities I downloaded the recording onto a PC where it played back OK at normal speed. I also loaded it into a video editor which showed it as a 30fps file with the appropriate number of frames.

In my log file when I set up the video format I get
{2020/09/12 21:53:12} Set up internal detect width=121 height=68
{2020/09/12 21:53:29} Change: video_width = 1920
{2020/09/12 21:53:29} Change: video_height = 1080
{2020/09/12 21:53:29} Change: video_fps = 30
{2020/09/12 21:53:29} Change: MP4Box_fps = 30
{2020/09/12 21:53:29} Change: image_width = 2592
{2020/09/12 21:53:29} Change: image_height = 1944

Just seen your comment about it seems OK from a command line capture. That indicates the basic recording by raspimjpeg is OK but maybe frames are getting dropped somehow when browser is operational. You can put %k on the annotation string which will add the frame number (0-29) for 30fps. This would help see if frames are getting dropped.

Another thought is whether you are using a separate storage device for the recordings rather than SD. If so then it is very desirable to use the boxing_path setting as described in the wiki. This allows th eraw h264 recording to be made to SD and then this is Boxed to the final destination as an mp4. The recording process is time critical whereas the Boxing is not.

Is there somewhere where you could post a 20 second bad mp4 for me to check out?

Skipher, 2020-09-12:
I've actually been checking the number of frames using %k in the annotation string.

I will say though that I was able to achieve 30fps ONE TIME on my Pi 3B+ at 720p after reinstalling everything several times, but I can no longer get 30 fps for some reason.

I am currently using a Pi 4, and the frame number gets to 15 and 16 in an alternating manner even though the fps is set to 30.

RichShumaker, 2020-09-12:
I am using s Pi4 booting from USB3 and an HQ camera. I will film a video today with an analog hands clock on my phone and use the %k for the frames.

Skipher, 2020-09-12:
I have new results to report. I've reinstalled everything on both my Pi 3B+ and 4 and have tested the same camera on both hardware.

Both 3B+ and 4 have the issue of streaming only 15-16 fps despite setting the fps to 30. They are both connected to a 5 GHz network.

When I manually record a video through the web interface, 3B+ can record at 720p 30fps and 1080p 23-24fps. It appears that at 1080p, the 3B+ is losing some frames. Is it not powerful enough?

With the Pi 4, when I manually record a video through the web interface, it can record at 720 22-23 fps and 1080p 15-16fps. I am confused because Pi 4 should be more powerful.

RichShumaker, 2020-09-13:
Testing done. This issue and Skipher are the same.
I thought mine was different but it is exhibiting the same exact issue.
Can we merge the issues?

Here are the results that have been put on my website.
http://richshumaker.com/raspi/fps/

The Files are not converting in the background to mp4 so I have been using the H264 files.
If you need me to convert them let me know.
Also if you have a good way to convert on a Win10 machine let me know as VLC is being a pain for me, but it still does the job just annoying as it keeps getting caught in loops.

Files I created
1080P 30fps From Browser
640P 90fps From Browser
All Browsers Turned Off - CA1-CA0 on the command line with all my browsers closed. I typically use my PC phone and iPad sometimes all at once.
I also put in PiVideo.h264 which was created with this command after turning off the camera in the browser
raspivid -t 30000 -w 1920 -h 1080 -fps 30 -b 2200000 -p 0,0,1920,1080 -a 512 -o pivideo.h264
I could not for the life of me get the %k annotation to work on the command line with -a "%k" or '%k' so clearly user error. This video has the total frames on screen not as useful as %k

The browser files and the CA1-CA0 Files have the issue.
They are all half framed - Meaning 30fps is 15fps and 90fps is 45fps.

The Command line file seemed good. but since I could not do the %k I am not 100% sure.

Please let us know what other testing needs to be done.

Skipher, 2020-09-13:
I went ahead and closed my issue.

Skipher, 2020-09-13:
I have another behavior to report:

On my 3B+ currently, I have set the fps to both 23 fps because this seems to be the minimum fps at 1080p at which it can record without losing frames.

As I've said before, when I open the web interface, the streaming fps is only 15 fps, and this was at a preview quality of 10.

For some reason, I decided to test quality = 100, and now the live preview/stream can reach frame number 22, which means that it is now also streaming at 23 fps. Unfortunately, when I go up to 30fps, the stream does not also reach 30 fps even though the quality is at 100. Although the stream now skips some frames due to the increased bandwidth over WiFi, the frame number eventually gets to 22: 0, 3, 7, 10, 12, 17, 19, 22, for example.

Skipher, 2020-09-13:
Quick update to my last test. Bringing the quality back down from 100 to 20 keeps the streaming fps at 23. Setting the quality to 100 in the first place seems to have "enabled" something.

RichShumaker, 2020-09-13:
> Quick update to my last test. Bringing the quality back down from 100 to 20 keeps the streaming fps at 23. Setting the quality to 100 in the first place seems to have "enabled" something.

This only changed the streaming correct? You video recordings stayed at 15fps?!?

Skipher, 2020-09-13:
Currently, my recording fps seems to be working properly. When I hit the button to record a video, the streaming fps actually does go up to 23. However, the streaming fps returns to 15-16 when the recording stops.

The max fps I can record at without losing frames is 23 fps at 1080p. I'm just not sure why the 3B+ is not capable of recording 1080p 30fps through the web interface but is fine using raspivid.

RichShumaker, 2020-09-13:
Thanks @Skipher as I was wondering what your FPS got to when recording. Since others are not discussing this issue and we just installed this must be something new(I think I installed a month ago). I don't know how to install previous versions BUT that would be my next step to see if an update changed something. Since our FPS is halved that leads me to think some type of buffer got doubled & as a result halved our fps(it is way too consistent at 15fps). I am glad that it is not on raspivid as that would have made me think kernel issue and I am not advanced enough to do much beyond simple stuff, which ain't Kernel coding.

Skipher, 2020-09-14:
I remember having this same issue with version 6.6.6 (if I remember the number correctly). I recently updated to 6.6.11 and decided to check if the issue was gone, but it apparently is not.

RichShumaker, 2020-09-14:
I need to learn how to roll back my version.

RichShumaker, 2020-09-14:
I did some additional testing. and I found something weird.
I filmed it except I did not put subtitles so I will need to describe what happened.
This was a stream capture of a Web Browser in vMix.

File for this commentary
http://richshumaker.com/raspi/fps/StreamCaptureForTiming.mp4
I started at 90fps 640 X 480(15fps shows on screen)
I switched to 1080 30p - screen format changes(white bar at bottom) but the fps stays at 15
I then switched to another format - 15pfs max still
Then back to 640 X 480 90fps(what I started with) again 15fps.
I then started a video record and that is when the fps went to 30 and I think it might have gone slightly higher but I might be seeing a ghost number.
Then I stop the recording - 30 is stuck on screen until the file is done saving and then back to the 15fps on screen.
The recorded file goes to 30fps

RichShumaker, 2020-09-15:
> I remember having this same issue with version 6.6.6 (if I remember the number correctly). I recently updated to 6.6.11 and decided to check if the issue was gone, but it apparently is not.

I remember getting mad a few years ago at RPi Cam Web Interface and stopping using it. I don't remember if it was this issue but it does feel familiar to me. I absolutely Love this software always have and I just hope I can roll it back or figure this glitch out.

RichShumaker, 2020-10-01:
I wanted to post this YouTube Live that I did that shows the HQ Camera with a Pi4 running from USB3 and hardwired into my network. It shows this issue both when recording and not.

https://youtu.be/y5IwNFpTvEI

hogwell, 2020-12-05:
I am also seeing the same behavior (%k not going beyond 15fps on the raspimjpeg website stream despite a setting of 25fps.)
(Running on a pi4 w/4GB)

Maybe this is a limit of how fast raspimjpeg can write the cam.jpg files.

Or it could be a limit of the GPU and the HQ camera?

RichShumaker, 2020-12-05:
> I am also seeing the same behavior (%k not going beyond 15fps on the raspimjpeg website stream despite a setting of 25fps.)
> (Running on a pi4 w/4GB)
> Or it could be a limit of the GPU and the HQ camera?

MjPEG from what I was told is not limited compared to the GPU ISP which is limited to 1080p 30fps 8 bit color.
MJPEG at 1%(which looks crazy) has almost no bandwidth and it seems that this issue is being caused somewhere else. I have not tried a clean install using a Pi 0/1/2/3 with a V1/V2 to see if I can or can not reproduce the issue.

hogwell, 2020-12-06:
I see there is a newer version of RaspiMJPEG that supports a new 'fps_divider' parameter.
I don't understand how to use this new option - maybe it addresses this issue on the HQ camera?

username7291, 2021-01-07:
I can confirm that rpi2/Buster/newest Version/HQ camera/FHD-24fps. I record a Video of a RF controlled clock, start frame 51:07 end 52:32 (MMSS). Means 1M25s, the Video lenght is shown as 1M17s(MediaInfo) and 1M18s(Mediaplayer/Browser). MP4Box put the h264 only in a Container(Its pretty slow even from RAM to SD[Class 10 UHS1] for that, about ffmpeg omx encoding mjpg->mkv/h264), so it should not change the Video.

username7291, 2021-01-08:
I made some more experiments. Its not the GPU Speed 250/400MHZ make no real difference. It seems the internal Motion Detection. If i disable it, start a manual Record i get a file that is 1M1s and the Clock shows 1M2s (within measurement error). The slowness of MP4Box is from linux/fat32/sync issue. With fat32-flush MP4Box takes 23s/100MB, ffmpeg takes 15s/100MB.

BETLOG, 2023-10-28:
It's now 2023 and I now also have this issue.
I have been running 4 pizeroW's with the same setup for several years, and while blowing off dust and reassessing heatsinking I recently made some update attempts to current raspbian (bookworm). When I realised that wasnt going to happen I fell back to fresh re-installs of Buster, with some config pasting/tweaking.

- Fresh Buster installs
- TF bootcode.bin, boot and rootfs on USB3-enclosed M.2 NGFF drives
- OV5647 75deg - https://www.aliexpress.com/item/32946093276.html - https://www.geeetech.com/wiki/images/8/81/OV5647_breif.pdf
- ~5.3V 3A PSUs - https://www.aliexpress.com/item/1005001759087720.html

The only things I have done different with my configs this time around is added some parameters to hopefully mitigate the regular hangs I get after temperatures spike for no apparent reason.

> /boot/cmdline.txt
```
# console=serial0,115200 console=tty1 root=PARTUUID=82cb65cc-82b9-44f7-91f1-36ec7e451b99 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
console=serial0,115200 console=tty1 root=PARTUUID=82cb65cc-82b9-44f7-91f1-36ec7e451b99 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait smsc95xx.turbo_mode=N
```

>/rootfs/etc/sysctl.conf (appended)
```
vm.min_free_kbytes = 16384
# vm.min_free_kbytes = 32768

```

Otherwise my config.txt is unchanged:
> /boot/config.txt
```
# betlog - 2023-10-24--13-27-30
# pizero W - 10 Buster - RPiCWI

#mandatory
start_x=1
gpu_mem=128

#stealth
#dtparam=act_led_trigger=default-on
#dtparam=act_led_trigger=none
#dtparam=act_led_activelow=on
#disable_camera_led=1

# preference
dtoverlay=disable-bt
dtparam=audio=off
```

...except for video_fps, which has always been 25 until now. That said; setting 25fps now still results in visible time compression.
> /rootfs/var/www/html/uconfig
```
annotation %Y-%M-%D--%h-%m-%s.%u %a
anno_background 1
anno_text_size 14
metering_mode matrix
exposure_mode off
white_balance off
width 1024
quality 13
video_width 1296
video_height 972
video_fps 42
video_buffer 3000
MP4Box 2
MP4Box_fps 42
image_width 1296
image_height 972
image_quality 13
image_path /var/www/html/media/%Y-%M-%D--%h-%m-%s.jpg
lapse_path /var/www/html/media/%Y-%M-%D--%h-%m-%s.%u.jpg
video_path /var/www/html/media/%Y-%M-%D--%h-%m-%s.mp4
motion_external 0
motion_noise 1009
motion_threshold 125
motion_image /var/www/html/motionMask-1296x972.pgm
motion_stopframes 300
initial_quant 18
encode_qp 18
```

Suddenly I'm seeing [video thats time compressed](https://drive.google.com/file/d/1lN5yJ-nmb9IdoMlD5PwmN4iCZRaRhwMs/view?usp=sharing), and people moving at high speed.

NOTE: all of these video have been recorded with the 'Record video start" button in the RPiCWI... interface. I recorded at least 20 seconds, then cropped each video to 20 seconds based on it's own timestamps using (for example):
```
ffmpeg -hide_banner -nostdin -loglevel fatal -stats
...
-filter_complex "0:v:0]trim=start=0.233:end=17.167,setpts=PTS-STARTPTS[v0]"
...
-c:v libx264 -qp 18
```

 Then compressed them for filesize, and briefly confirmed (visually watching the timestamps click over a bit too fast in a video or two, and in smpayer playlist duration timestamps) that the time compression issue is unchanged by the compression.
[image]
The microsecond timestamp and the frame/framerate is visible in the cam overlaytext - in the videos.

I realise doing this at night with no people walking by isnt ideal, but I don't think it alters the results.

Also: Feature request:  option for different daytime and nighttime framerates.... <11Hz picks up a lot of light!

Whats going on?
Has a recent update to the Buster repo broken it? I see some of those are dated quite recent, and it's pretty standard for older OS versions to be weirdly broken by changes made and tested primarily in a new OS version.

P.S: +1 for undertaking the arduous journey toward libcamera integration into Glorious and Without Peer Master Race RPiCWI.

roberttidey, 2023-10-28:
The most common reason for this is frame dropping.

To see if this is the case make a recording ( I suggest using 25 fps) with %k included in the annotation string. This will number the frames (0-24) for 25 fps. You can then step through recording using a video editor and check whether all frames remain in sequence.

If you get frame drop then most likely reason is the speed of storage. This is particularly the case if storage is linked to some other medium.