silvanmelchior/RPi_Cam_Web_Interface #585: Accessing the mjpeg stream

riccardopinosio, 2020-10-05:
Hi,

I am trying to access the mjpeg stream from rpi cam web interface but can't manage. The cam is attached to a pi 0 and i install rpicam from diet-pi optimized software selection, which exposes the frontend at http://camera_ip/rpicam.
When I navigate to that url I see the stream embedded in the webpage, which gets updated allright.
However when I go to /rpicam/cam_pic_new.php the browser freezes. Similarly, if I try to stream it on VLC it hangs and doesn't show the stream.
I am however able to access the most recent picture by navigating to /cam_pic.php.
Any idea what might be causing this issue?

roberttidey, 2020-10-05:
try using
/rpicam/cam_pic_new.php?pDelay=40000

where pDelay is refresh period in microseconds

If that doesn't help tell me what version of the software you are using.

riccardopinosio, 2020-10-06:
Hi @roberttidey,

That didn't help, still the stream doesn't appear. I am using RPI cam installed from the optimized software list of DietPi at: https://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p48,
not sure what version that is.
I tried to install RPIcam on dietpi directly (not using the optimized software from dietpi but with the install instructions for rpi cam web), but got a different error.
Gonna try with raspbian lite and see if that works.

riccardopinosio, 2020-10-06:
update: I am able to make it work when I am on raspbian lite and install using the install guide.
I guess this is an issue related to dietpi, hence, I am closing this issue an opening one there.

closed 2020-10-06 by riccardopinosio