silvanmelchior/RPi_Cam_Web_Interface #586: Execute python with userbutton

Bond246, 2020-10-12:
Hello,

i'm trying to implement some userbuttons to control a BrightPi Module. 
BrightPi is controlled via I2C and my scripts talk via python.

To have everything i need in one script i wrote a "led_button.py" and to make it work with userbuttons i wrote some one-line shell scripts like this:
```
#!/bin/bash
python3 /var/www/macros/led_button.py --white
```

permissions of both files are:
```
-rwxrwxr-x 1 www-data www-data led_button.py
-rwxrwxr-x 1 www-data www-data led-taglicht.sh
```

My userbuttons looks like that:
```
Tag-Light,led-taglicht.sh,btn btn-info btn-lg,style="width:10%"
```

When i execute led-taglicht.sh from shell everythin works fine.
When i execute it from userbuttons nothing happens. scheduleLog has the exec-line:
```
{2020/10/12 18:49:39} Executing macro /var/www/macros/led-taglicht.sh &
```
But nothing is happening.
Even using the full pyhton3-path /usr/bin/python3 was not working.

Any ideas what the problem may be?

thymjan, 2020-10-12:
Nearly had same problems. Try to execute the stop.sh script in a shell and afterwards start it again with the start.sh script. Keep an eye on the shell and push your buttons, maybe you get some useful error messages there as I did.

thymjan, 2020-10-12:
Maybe the www-data user needs i2c rights:
sudo apt install i2c-tools
sudo adduser www-data i2c
sudo reboot

Bond246, 2020-10-12:
Haha!!!
That helped!!!

I had to add the user www-data to the i2c-group.

How can i see on shell what is going on if i push a button in RPi-Cam?

closed 2020-10-12 by Bond246