silvanmelchior/RPi_Cam_Web_Interface #587: Difficulty in adding second Pi camera...

Bromley47, 2020-11-02:
I have two RPi Zero W's with Pi V2 cameras...I can see both on separate web pages no problem But unable to view on one page. 

I have followed the instructions as follows:

Made a mutliview.json file in /home/pi/RPi_Cam_Web_Interface/www and copied the text from multiview.jsonD and pasted into the new file. Then changed the url for the '0' and 'to the ones for each camera.

Used ip/multivied.html to try and show both cameras but says 404 not found. I have also commented out the extra two cameras (192.168.0.4 and 5) as described (not sure if this matters or not) Shown as cameras 192.168.0.2 and 0.3. 

{"hosts":["http://192.168.1.xxx/","http://192.168.1.xxx/","http://192.168.0.4/","http://192.168.0.5/"],"delays":[40000,40000,40000,40000]}

<!DOCTYPE html>
<html>
  <head>
    <title>RPi Cam Multiview</title>
    <script language="javascript">
		var mjpeg_img0,mjpeg_img1,mjpeg_img2;
 
		function load_img0() {mjpeg_img0.src = "multiview.php?time=" + new Date().getTime() + "&pHost=0";}
		function load_img1() {mjpeg_img1.src = "multiview.php?time=" + new Date().getTime() + "&pHost=1";}
		function load_img2() {mjpeg_img2.src = "multiview.php?time=" + new Date().getTime() + "&pHost=2";}
		function load_img3() {mjpeg_img3.src = "multiview.php?time=" + new Date().getTime() + "&pHost=3";}
		function init() {
			mjpeg_img0 = document.getElementById("mjpeg_dest0");
			if(mjpeg_img0) mjpeg_img0.onload = load_img0();
			mjpeg_img1 = document.getElementById("mjpeg_dest1");
			if(mjpeg_img1) mjpeg_img1.onload = load_img1();
			mjpeg_img2 = document.getElementById("mjpeg_dest2");
			if(mjpeg_img2) mjpeg_img2.onload = load_img2();
			mjpeg_img3 = document.getElementById("mjpeg_dest3");
			if(mjpeg_img3) mjpeg_img3.onload = load_img3();
		}
	</script>
  </head>
  <body onload="setTimeout('init();', 100);">
      <img id="mjpeg_dest0"  style="width: 49%"/>
      <img id="mjpeg_dest1"  style="width: 49%"/>
      <---<img id="mjpeg_dest2"/>-->  style="width: 49%"/>
      <---<img id="mjpeg_dest3"/>-->  style="width: 49%"/>
 </body>
</html>

I have also played around with iSpy to try and download the images from both cameras because I believe that a second camera on RPi_Cam_Web interface does not save the images . Is this correct.

Could not get a second camera to load on iSpy either so maybe a dead end there.

Any help appreciated.

Bromley47, 2020-11-03:
Sorry this is rest of html file which got missed off...
</head>
  <body onload="setTimeout('init();', 100);">
      <img id="mjpeg_dest0"  style="width: 49%"/>
      <img id="mjpeg_dest1"  style="width: 49%"/>
      <---<img id="mjpeg_dest2"/>-->  style="width: 49%"/>
      <---<img id="mjpeg_dest3"/>-->  style="width: 49%"/>
 </body>
</html>

Bromley47, 2020-11-03:
</head>
  <body onload="setTimeout('init();', 100);">
      <img id="mjpeg_dest0"  style="width: 49%"/>
      <img id="mjpeg_dest1"  style="width: 49%"/>
      <---<img id="mjpeg_dest2"/>-->  style="width: 49%"/>
      <---<img id="mjpeg_dest3"/>-->  style="width: 49%"/>
 </body>
</html>

Bromley47, 2020-11-03:
Cannot seem to copy and paste that part...

roberttidey, 2020-11-06:
I am checking this out.

multiview just gives a separate view integrating the previews from up to 4 cameras. It does not have any effect on the settings or controls of those cameras which remain with their normal controls.

Bromley47, 2020-11-06:
Thanks for the reply Robert - what I would like to do is use a single view showing two cameras and two separate views showing each camera enabling me to download the images etc. I can get the two separate views on separate web pages but not the single view showing both. I want to be able to use this on separate screen in a different part of the house than the PC is.

Hope you can point me in the right direction. One question - when you say access the multiview with ip/multiview.html does it matter which ip address is used?

Bromley47, 2020-11-15:
Robert - did you have any  luck in sorting what the problem could be with multi view?

roberttidey, 2020-11-16:
I hope to get some time to check this out in a couple of days time.

BETLOG, 2020-12-27:
Surely somebody has already done this and can offer a sample as a template requiring edits?

roberttidey, 2020-12-27:
I haven't seen this yet. One fairly easy approach is to start with multiview and add a link each on each image to preview.php on the relevant host.

BETLOG, 2020-12-28:
> start with multiview

Unfortunately to those of who develop for giggles rather than money phrases like this are not self explanatory :)
The generalised how-to I read elsewhere (probably the wiki) is similarly not that useful to many of us. An actual code example is needed for copy-paste-modify implementation.

pishotta, 2020-12-28:
It is not for giggles that Robert does this. He is a Saint.

roberttidey, 2021-01-01:
I am looking at improving the multiview example so that it can redirect through to the individual camera servers to allow looking at the captures etc.

I am busy with other stuff at the moment so may take a few days to get round to this.