silvanmelchior/RPi_Cam_Web_Interface #589: Long exposures for nighttime stills?

2ahunter, 2020-11-06:
Is there a way to take a still picture using exposure times longer than one second?  I can do this with picamera by setting a low framerate but i can't figure out how to do this with the web interface. I have the HQ camera and want to do exposures on the order of 1-20 seconds.

roberttidey, 2020-11-06:
Currently you can only set the fps as an integer value so the slowest is 1 fps.

I am looking at making a small change so you can have a divider on this to allow much slower rates. E.g with a divider of 20 and 1 fps then you would get 1 frame per 20 seconds.

I will update here if I get this working.

roberttidey, 2020-11-06:
I have added support for fps divider in latest version.

2ahunter, 2020-11-06:
Thank you!  If this works it will make my project much easier.  I have pulled the repository, where do I find the divider? I see the one in the image preview window, but that was there before.

2ahunter, 2020-11-06:
Well, the changes are seen in the preview window--image is clearly exposed longer, however, the captured image remains at 0.996 seconds.  The thumbnail previews shows the "correct" exposure but the captured image always is capped at just under 1 sec.

roberttidey, 2020-11-06:
There is a fps_divider function in the first part of camera settings where the resolution and fps is set.

2ahunter, 2020-11-06:
Yes, I see the divider function in the resolution box. It seems to work. However, the shutter speed field doesn't change after values greater than 1000000, i.e., 1 second. Is there a limitation in the code that limits the value of that field?

roberttidey, 2020-11-07:
I am checking into that further.

pr-boh, 2022-01-30:
I am engaged in taking photos from the sky by night (Astrophotography with the Raspberry Pi Camera). I mounted a HQ Raspicam on my telescope. Handling is done with this software and this is very convinient. But sometimes I need long exposure > 1 second. 
Therefore I agree with 2ahunter. It would be nice having the possibilty taking pictures with exposure > 1s .
[image]

roberttidey, 2022-01-30:
Although the shutter speed control in the web interface implies maximum value is 330000 uSec (0.33 second)  it will pass through any entered value to the camera. I think the maximum is 6000000  6 seconds for v1 camera and maybe much more for HQ camera (200 aeconds?)

I will make the limit clearer and widen the entry box on the web interface to make entry easier.

pr-boh, 2022-01-31:
I tested different shutter values last weekend from 300000 to 6000000.  Equipment: Raspberry Pi 3B+ and HQ camera modul. OS was bullseye.  And as 2ahunter mentioned before there ist no change in picture above values from 1000000. With astronomy programs kstars and ekos exposure times > 1s are possible and you can see corresponding changes in picture.

roberttidey, 2022-01-31:
It is definitely sending the longer values into the MMAL camera control software.

But I might have found a time-out on waiting for something to complete which could affect things. Under investigation.

roberttidey, 2022-01-31:
It doesn't look like the timeout should affect things as it is set to a high value. It is set in the /etc/raspimjpeg config file (can't be changed from web interface). If you set it to 0 then it disables it so that might be worth trying.

The other thing to try is to set exposure compensation and white balance to off.

pr-boh, 2022-01-31:
Thanks, I will try!

pr-boh, 2022-02-01:
Hi, no success so far!

HamHamFonFon, 2022-08-03:
I am interested with long exposure too for astrophotography. Is it possible to add a little paragraph in wiki to how take photo with "n" seconds with the explanation below ?