silvanmelchior/RPi_Cam_Web_Interface #591: Keep losing connection to Camera and requires a power reboot

Bromley47, 2020-11-19:
I have two Pi MMAL cameras on separate Pi Zero W boards - each board on a  separate system with separate IP addresses.

One works fine, the other I keep losing the WiFi connection to it and requires a power reset to get it running again - which is a bit inconvenient as it is sited remote from the house (but with a good WiFi signal).

Would appreciate any help to solve this one.

roberttidey, 2020-11-21:
Have you tried turning the camera off on the Pi that loses wifi connection to see if it still suffers a lost connection? That would establish whether it was associated with camera activity.

It is worth double checking your power arrangements to make sure there is a good 5V at the Pi itself. If using a USB power feed then some cables can have excessive resistance causing voltage drop. This is exacerbated by using the camera which draws extra current.

Bromley47, 2020-11-21:
Thanks for the advice Robert. I will try that as one has a longish power cable.

Any luck with the multi-view?

roberttidey, 2020-11-21:
I have set up my multiview in standard way and it seems to work Ok. I will be doing some more testing and will update when I have done that.

Bromley47, 2020-11-21:
OK Robert - one question - if I have two cams what IP address do I use to view to show the multi view or doesn't it matter - either?

roberttidey, 2020-11-21:
On one of the pi cameras you configure multiview.json (based on multiview.jsonD) to have the ip addresses of the cameras you want monitored.

When you then access multiview.html on the same raspberry pi then it then accesses each image stream using multiview.php to fill the image areas of the html with the stream from each of those cameras.

By default it is set up for 4 cameras but that can be changed to a different number by changing the html and the json file.

So if only two are needed then one cuts the html down by deleting two of the img components in the body, deleting two of the function load_img parts in the javascript in the same file and by only putting two hosts in the json file.

Bromley47, 2020-11-21:
I am trying to access the multiview using http://192.168.1.xxx/html/multiview.html. Is this correct or do I not need the html after the IP address. If i get rid of that its says:-

Not Found
The requested URL was not found on this server.
Apache/2.4.38 (Raspbian) Server at 192.168.1.xxx Port 80

This is the entry in multiview.json:-

{"hosts":["http://192.168.1.xxx/html/","http://192.168.1.xxx/html/"],"delays":[40000,40000,]}

xxx being the IP address of each camera.

Also deleted load_img parts in mutliview.html on ONE camera only.

function load_img0() {mjpeg_img0.src = "multiview.php?time=" + new Date().getTime() + "&pHost=0";}
		function load_img1() {mjpeg_img1.src = "multiview.php?time=" + new Date().getTime() + "&pHost=1";}
		
		
		function init() {
			mjpeg_img0 = document.getElementById("mjpeg_dest0");
			if(mjpeg_img0) mjpeg_img0.onload = load_img0();
			mjpeg_img1 = document.getElementById("mjpeg_dest1");
			if(mjpeg_img1) mjpeg_img1.onload = load_img1();
			mjpeg_img2 = document.getElementById("mjpeg_dest2");
			if(mjpeg_img2) mjpeg_img2.onload = load_img2();
			mjpeg_img3 = document.getElementById("mjpeg_dest3");
			if(mjpeg_img3) mjpeg_img3.onload = load_img3();

And to make sure you have it all this is how I commented out the two extra cameras:-

body onload="setTimeout('init();', 100);">
      <img id="mjpeg_dest0"  style="width: 49%"/>
      <img id="mjpeg_dest1"  style="width: 49%"/>
      <---<img id="mjpeg_dest2"/>-->  style="width: 49%"/>
      <---<img id="mjpeg_dest3"/>-->  style="width: 49%"/>

Would welcome your comments on the above and thank you for your assistance.

Bromley47, 2020-11-21:
Sorry - again its not pasted in:-

<body onload="setTimeout('init();', 100);">
      <img id="mjpeg_dest0"  style="width: 49%"/>
      <img id="mjpeg_dest1"  style="width: 49%"/>
      <---<img id="mjpeg_dest2"/>-->  style="width: 49%"/>
      <---<img id="mjpeg_dest3"/>-->  style="width: 49%"/>

Bromley47, 2020-11-21:
Cannot do that - most frustrating..!!

roberttidey, 2020-11-21:
Ok. Here is a step by step approach starting from the standard supplied files.

Copy multiview.jsonD to multiview.json in the installed html folder on one pi (e.g. /var/www/html)

sudo cp multiview.jsonD multiview.json

Edit and save the multiview.json file to have the same ip address in each of the 4 positions. E.g. if that pi was 192.168.0.20 then change all 4 ip addresses to 192.168.0.20

sudo nano multiview.json

Now access the multiview.html from a browser 

http://192.168.0.20/html/multiview.html

You should now see 4 image preview streams on the same page with the same content.

Now edit the multiview.json file to have the ip addresses of the other cameras. If there are less than 4 then just replicate one of them.

Test again and you should see the different cameras again with any replication.

Now edit the multiview.html to remove any redundant views. For simplicity I would take out both the 
		function load_imgx()
and the corresponding
			mjpeg_imgx = document.getElementById("mjpeg_destx");
			if(mjpeg_imgx) mjpeg_imgx.onload = load_imgx();
lines

Bromley47, 2020-11-22:
Thank you Robert - worked a treat - maybe these instructions should be included in the WiKi instructions...just a suggestion.

Again cheers