silvanmelchior/RPi_Cam_Web_Interface #593: New scheduler command: turn camera off and on, and new settings for off banner (text or picture)

sensboston, 2020-11-20:
Hi. First of all, I wanna say "Thanks!" for this great project! I'm using RPi_Cam_Web_Interface for a backyard/driveway monitoring and it works pretty good, seamlessly for the years. 

Now I'm working on ["Bird feeder cam"](http://birds.servepics.com/) tiny project; work still in progress (I wanna add nice gallery for the captured pics & videos) but you can check it live. If someone interested, I can PR my changes to the mainline or forked repo.

But I'd really like ability to completely turn off camera in evening (and show banner or stub picture instead live stream), and turn it on in morning. I know, this task can be done by calling external scripts and modifying landing page but I think this feature should be added to the huge features set of this perfect application.

I'm not an expert in web programming, so, implementing this simple feature by myself will take a long (longer) time. @silvanmelchior, I believe, this feature implementation will take not more then 30 minutes of your time :wink: 

By the way, might be I've missed something and this feature can be implemented without changes; if you know how, please let me know (with detailed example please if possible)

Thanks!

roberttidey, 2020-11-20:
I think you can do something similar already.

When you stop the camera from the web interface it displays an unavailable message. This is actually just an image file called unavailable.jpg located in web folder. You can replace this with a different image but with same name to display something different.

There is an available command (ru) to perform the same action on the camera (ru 1  to start camera, ru 0 to stop it). These should work in the scheduler.

sensboston, 2020-11-21:
@roberttidey, thanks for suggestions, I'll try tomorrow and post an update here.

BTW, I do have another suggestion/question (but don't wanna open a new issue): probably, for the most of RPCWI users without **nginx** or **Apache** administrative experience, will be very helpful to post suggested "view only" configuration files for these servers. Yesterday I've spent a few hours by googling and trying to configure **nginx** for "view only" behavior (by default, RPCWI is wide-open for DDoS attacks). What do you think?

P.S. I'm curious, how safe is RPCWI at all (being open to the internet)?

roberttidey, 2020-11-21:
RPCWI is pretty safe in my view.

First if you open it up to internet you would normally use port forwarding. I normally use a non-standard high port number for this. Although this won't deter port scanners that look across whole range it is an extra hurdle to get over.

Second I would always use a password when opening up to internet.

Third the web server files run under user www-data which has limited user privileges. It is wise to make sure that any custom scripts (macros) installed are as limited in scope as they can be. There have been some improvements to url handling over the years to try to prevent getting enhanced access by creating complex urls with embedded commands.

DDoS is a different matter. I'll check this out a bit further.

sensboston, 2020-11-21:
@roberttidey, I mean a lil bit different thing: if you wanna to open your cam to world (my project - see first post here - is good example), by default some scripts (without editing nginx configuration) are still available for execution, even you specified min.php as landing page. I.e. some weird guys still can execute index.php, start recording or run script to overload your sdcard by captured pics etc.

For example (but I'm unsure about correctness - I'm not good enough in **nginx** configuration, even in Linux at all 😉 ) here is my **nginx** config: [rpicam.zip](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/files/5578218/rpicam.zip) But I spent an hours to google "how to".

Might be, will be good to include similar stuff to the project and wiki? Just an idea, don't judge...

sensboston, 2020-11-23:
@roberttidey, I tried commands **ru 1** & **ru 0**; ru 1 worked (at least, I put ru 1 for 9 a.m. and stopped camera before going to bed, when I checked on next morning, camera was on, but ru 0 not worked for me. I set this command execution at 17 (5 p.m), checked at 6 p.m. and camera still was on (see pic of my config)
[image]

roberttidey, 2020-11-23:
First try using the commands manually from a command window.

camera off
sudo echo 'ru 0' >/var/www/html/FIFO

camera on
sudo echo 'ru 1' >/var/www/html/FIFO

They should take effect immediately.

Second check the scheduler log for what is logged around the scheduled time when the ru commands should run.

sensboston, 2020-11-25:
@roberttidey, thanks! It was my fault: I forgot to "start" scheduler 😉 By the way, "start" button for scheduler, from my sight, is "unnecessary entity" (but I don't insist).

Sorry for asking, but could you please teach me, how to program (inscheduler) RPCWI to make a shot for detected motion, with some interval(s)? I trying to automate my "Bird feeder cam" to get a nice cam shots/pics without human involved.

BTW, later (I'm working on "read-only" php screenshot gallery now - it will take some time 'cause I'm pretty new to php and web development at all) I can issue a PR to the project; I believe, nice cheap open source "bird feeder cam" can be popular to the "regular customers", not just enthusiast and developers.

roberttidey, 2020-11-25:
In scheduler you can use the motion start and motion stop fields to determine what sequence of commands you need.

A sequence is entered just by using a ';' between each command.

For example if you want a time lapse sequence to start when motion is detected then just use tl 1 to start it and tl 0 in the motion stop to stop it. You can also change the time lapse interval with the tv command.

If you want a very short interval then it is best to use video capture with a slow frame rate, e.g. 4 fps would give you 250ms between each frame in the video.

The other technique that can be used to provide more complex actions is to create macro (shell scripts) that are fired when motion starts or stops. These shell scripts can perform internal commands by piping them into the command FIFO and other external actions as required.

The reason for the stop start of the scheduler is that there is actually 2 parts to the scheduler. There is the user interface you see in the browser but there is also an instance normally running in the background performing the tasks. The background piece will normally refresh its settings whenever new ones are saved from the browser but there are some instances when this wouldn't happen. Also if a new version of the scheduler is copied in to a running system then the background one would still keep running the old version. Stopping and starting provides a way of making sure the current one is running.

sensboston, 2020-11-25:
Thanks for help and explanation!

closed 2020-11-25 by sensboston