silvanmelchior/RPi_Cam_Web_Interface #594: Space after semicolon in scheduler causes second command to fail

ras07, 2020-12-01:
The scheduler supports multiple commands separated by a semicolon. However if the semicolon is followed by a space, e.g.:
`em auto; tl 1`
then the second command doesn't get executed. The log will say something like:
```
[2020/12/01 12:07:00] New period detected 8
[2020/12/01 12:07:00] Send em auto
{2020/12/01 12:07:00} Change: exposure_mode = auto
[2020/12/01 12:07:02] Send  tl 1
[2020/12/01 12:09:39] Scheduled management tasks. Next at 1606847079
```

(Note the two spaces between **Send** and **tl 1** on the penultimate line)
Technically this is user error (the instructions do say to separate commands by a semicolon, not a semicolon and a space), but it's bitten me several times and it's a hard one to diagnose. It would be more user-friendly if the scheduler ignored after a semicolon.

roberttidey, 2020-12-01:
I have updated latest version to handle this

ras07, 2020-12-01:
Great - Thanks!

closed 2020-12-01 by ras07