silvanmelchior/RPi_Cam_Web_Interface #595: Record for a set period on motion detect

Woody565, 2020-12-16:
Hi,

I am running the latest RPI Cam Web Interface and have it setup to record on motion which it does very well.

I would like to be able to set the camera to record for 5 minutes once motion has been detected and not stop once the motion has stopped.

Is this possible at all please?

roberttidey, 2020-12-16:
Yes.

The camera start command ca 1 can take an additional parameter which is the number of seconds to record for.

So ca 1 300 will perform a recording of 5 minutes.

If you put that in the motion start command and do not put anything in the motion stop command then that should achieve what you want.

Woody565, 2020-12-16:
Thanks Robert, 

worked perfectly