silvanmelchior/RPi_Cam_Web_Interface #597: I'd like to change the filenames

BETLOG, 2020-12-27:
Will I break anything internally if I edit:
/var/www/html/uconfig

with something like:
image_path /var/www/html/media/%Y-%M-%D--%h-%m-%s.jpg
lapse_path /var/www/html/media/%Y-%M-%D--%h-%m-%s.%u.jpg #note the additional %u
video_path /var/www/html/media/%Y-%M-%D--%h-%m-%s.mp4

Also:
annotation Cam1 %Y-%M-%D--%h-%m-%s
contains a camera name
is there a %variable substitution for this? I'd like to prepend it to my filenames.
ie: relevant data  like:
%CAMERANAME_%Y-%M-%D--%h-%m-%s.*
or
%HOSTNAME_%Y-%M-%D--%h-%m-%s.*

roberttidey, 2020-12-27:
Name changes should be fine (obviously without the #comment on end of lapse_path

There is no substitution for host or cameraname at present as I thought they could be hard wired into the names but I'll look into it.

BETLOG, 2020-12-28:
Given that %a can be anything; the second part is essentially solved by:
https://gist.github.com/BETLOG/7d161ad900dd957de5996e1a49e264a6
which i just added to the wiki

BETLOG, 2020-12-31:
As you indicate in the wiki, everything that we might ask for like I have above is accessible via %a, so I'll just close this.

It might however be useful to have %a1 %a2 etc so it would be possible to use more than one.

closed 2020-12-31 by BETLOG