silvanmelchior/RPi_Cam_Web_Interface #598: Video cropped / zoomed in

SignumPL, 2020-12-30:
When I take a picture using web UI it looks as I expect, like this:
[image]
However video stream area is smaller, looks cropped / zoomed in:
[image]

I didn't find any relevant setting for this behavior. How to set video stream to see entire area?

BETLOG, 2020-12-31:
note the difference in yours (and not in mine) of 

Video res:

Image res:

[image]

Mine are the same because the only thing I use image capture for is capturing images to edit motion masks from, so i leave them set to the same resolution.

SignumPL, 2020-12-31:
Ok when I set both resolutions to 1296x972 it works. But I don't understand why video is cropped when video resolution is set to 1920x1080.

BETLOG, 2020-12-31:
Well the full resolution of the sensor is probably generally preferred for image capture, but because of various overheads, not the least of which will be motion detection itself, it's probably best ot keep video resolutions to certain high-performance resolutions.
1296x972 for example is the highest res that still uses '2x2 binning', which happens at a very low level, therefore less processing overheads..... so i imagine the same logic applies to all of the standard resolution selections in the RPiCamWeb interface.

ctrl-f this for 'binning'

https://picamera.readthedocs.io/en/latest/fov.html#sensor-modes

xxLeoxx93, 2021-11-23:
For me it's also zoomed in, but setting both to 1296x972 doesn't help at all.
Anyone any Idea? I use the 5mp IR Cut Filter camera module.