silvanmelchior/RPi_Cam_Web_Interface #599: stretch install fails

BETLOG, 2020-12-31:
installing on stretch-lite becasue of overheating/lockups on buster....and the install auto-fails due to version discrepancies:

git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git
cd RPi_Cam_Web_Interface
cd bin
mv raspimjpeg raspimjpeg-buster
mv raspimjpeg-stretch raspimjpeg
cd ..
./install.sh
 Reading package lists... Done
 Building dependency tree       
 Reading state information... Done
 E: Unable to locate package php7.3
 E: Couldn't find any package by glob 'php7.3'
 E: Couldn't find any package by regex 'php7.3'
 E: Unable to locate package php7.3-cli
 E: Couldn't find any package by glob 'php7.3-cli'
 E: Couldn't find any package by regex 'php7.3-cli'
 E: Unable to locate package libapache2-mod-php7.3
 E: Couldn't find any package by glob 'libapache2-mod-php7.3'
 E: Couldn't find any package by regex 'libapache2-mod-php7.3'

sudo apt search php7
shows numerous 7.0 candidates, but installing them and rerunning ./install.sh does not allow the RPiCam install to continue

Notes of install prior to this problem stopping the install

```
sudo ddrescue -D --force /home/user/downloads/ISOs/pizero/2019-04-08-raspbian-stretch-lite.img /dev/sdf

# MOUNT SDCARD - if permission denied then card is fucked?

#     COPY ssh and wpa_supplicant TO SDCARD 
cp -a /home/user/documents/settings/pi/pizerocam1/boot/* /media/user/boot
ls /media/user/boot
#     COPY hostname TO SDCARD 
sudo cp -a /home/user/documents/settings/pi/pizerocam1/etc/hostname /media/user/rootfs/etc
cat /media/user/rootfs/etc/hostname

ssh-keygen -f "/home/user/.ssh/known_hosts" -R "192.168.1.61" # pizerocam1

sudo umount /media/user/boot
sudo umount /media/user/rootfs

#     REMOVE SDCARD AND BOOT PI
# wait 3 minutes for initial setup
ssh pizerocam1

default passswd:
raspberry 

passwd

sudo raspi-config
#     localisation options
#         locale
#             en_AU.UTF-8
#                 default=en_AU.UTF-8
#             timezone
#                 australia - brisbane
#             wifi country - AU
#     interfacing
#                 camera
#     advanced
#             expand filesystem 
# update raspi-config

exit # ssh pizerocam1

scp /home/user/.ssh/id_rsa.pub pizerocam1:/tmp
ssh pizerocam1 "ssh-keygen -C $(whoami)@$(hostname)-$(date +%Y-%m-%d--%H-%M-%S); cat /tmp/id_rsa.pub >>/home/pi/.ssh/authorized_keys"
# password from keepassX

ssh pizerocam1
# should be passwordless now
sudo apt update & sudo apt-get dist-upgrade -y

sudo reboot

ssh pizerocam1

sudo apt install -y git munin-node bc

sudo nano /etc/munin/munin-node.conf
allow ^192\.168\.1\.[0-9]{1,3}$
allow ^::ffff:192\.168\.1\.[0-9]{1,3}$
sudo service munin-node restart

git clone https://github.com/silvanmelchior/RPi_Cam_Web_Interface.git
cd RPi_Cam_Web_Interface
# NOTE: ONLY DO THIS IF YOU ARE RUNNING STRETCH OR EARLIER!
cd bin
mv raspimjpeg raspimjpeg-buster
mv raspimjpeg-stretch raspimjpeg
cd ..

./install.sh
```

roberttidey, 2020-12-31:
Buster should not in itself cause heating problems compared to stretch.

However, when installing with stretch 2 things are necessary.

1. Stretch does not have the same version binary libraries so the default raspimjpeg binary will not run. There is a version compiled for stretch compatibility in the bin install folder and this should be used to replace the default raspimjpeg e.g.
mv raspimjpeg raspimjpeg-buster; mv raspimjpeg-stretch raspimjpeg
2. Do not try to install a different php version for stretch. Instead set the compatible php version in the initial install.sh dialog (7.0 for stretch).

BETLOG, 2020-12-31:
1. did that
2. Sounds obvious now that you say it, will try it next time I set up to do it again.
You probably should add that to the wiki. Or do some OS version-based substitutions in the install script.

roberttidey, 2020-12-31:
I have not put anything specific in the install for stretch as most people should be using buster. The stretch version was added just as a workaround for users who didn't want to upgrade the os.

BETLOG, 2020-12-31:
ok, well the process for the user here is:
- burn and preliminary setup of OS
- run .install.sh, accept defaults
- get error
- remember that version field was present in ./install.sh options
- hopefully know apt search exists, do search to discover correct version number
- install the 3 failed items (lite doesnt have them pre-installed, dont know about regular)
- re-run ./install and enter new value

It'd be nice if the install script did some cat /etc/release, which, and apt install instead of making the user figure it out. People will always use older version to test problems against current releases.

closed 2021-01-01 by BETLOG