silvanmelchior/RPi_Cam_Web_Interface #601: mycam@

BETLOG, 2020-12-31:
Should mycam@ really be part of the page title?
Without checking I'm thinking it's hardcoded somewhere outside the gui or usual configuration files.

Presumably this is [another reason to need %cameraname, %hostname, etc variables](https://github.com/silvanmelchior/RPi_Cam_Web_Interface/issues/597)

[image]

roberttidey, 2020-12-31:
This is currently defined at the top of config.php

If I get a chance I'll see if I can move that out somewhere; maybe into /etc/raspimjpeg

BETLOG, 2021-01-01:
As a non-essential variable presumably. I would argue that it is not required. 

It isn't the username so it's not an ssh address.

It isn't the camera name.

It isnt an html address.

On my system it has no valid meaning whatsoever. So it's a little confusing to see it.