silvanmelchior/RPi_Cam_Web_Interface #602: No videos/images saved

username7291, 2021-01-01:
disabling thumbnail generation with "thumb_gen" makes the web interface dont find any files. I do not want thumbnails in the same dir like videos, but there is no sepeate thumbnail path setting.  Videos are there with Samba.
By the way the path settings are general a huge issue for usual users. Dont use the system partition is common, FAT32 is common. But its very difficult to change path to another drive or this thing with fstab and permission rights/owner. A better Manual about that would be a good idea.

roberttidey, 2021-01-01:
Thumbnails must be present for major parts of the system to work, e.g. preview and download captures, purging etc. This is because they are used as the index of what files are present and give a link to where the associated capture files are. Turning off thumbnails is a fairly rare requirement for those who are just configuring to do captures and not really using the web interface. 

Thumbnails are always stored in the root media folder. However, videos, images and time lapse files can be stored in completely different folders. For example, you can have video, images, and lapses as separate subfolders under the main media folder. This is achieved by just adjusting the various path definitions in the config file (/etc/raspimjpeg). These can even be on physically different volumes by using standard linux linking. These folders can even be extended into further subfolders by putting variables into the path definitions (e.g you can have monthly subfolders under a videos subfolder).

I recommend that any subfolders for these captures are logically kept under media and if you want them physically elsewhere then use linking. Although it is possible to use completely different paths other changes become necessary to avoid breaking stuff.

Likewise if you want to use remote volumes e.g. over a network for captures then I recommend that the thumbnails are still kept local and links are used to point to the remote volume. There is a facility called boxing folder (see wiki) which is particularly useful when using network shares as this does local captures during the video recording and then subsequently moves the recorded file afterwards. This avoids network delays causing recording problems.

I think the most common usage is with local SD storage. I agree that the wiki section on using other locations can be improved. Anybody can have improve the wiki.

username7291, 2021-01-03:
>However, videos, images and time lapse files can be stored in completely different folders

Thats a way, but, if i use samba to view the movies (Mediaplayer batch to view a whole folder) and then delete the files. The GUI would see the preview images and still think there are existent movies or?

With rpi people had sd corruption issues, cause its more secure to use a second partition or better a different storage where not is the system on it... FAT32 is the choice if the system brokes and you easy want access the files.

Another thing, if i try change Shutter Speed, the GUI get unresponive, camera image brokes and i cant change it back, even after reboot. I have to delete the entry from the config file. It was 600 on the HQ Camera.

roberttidey, 2021-01-03:
That's why there are delete facilities in the web interface. Also purge rules in scheduler can automate the process. Over my 5 cameras I have never had to manually delete a file over the last 4 years.

Likewise I have never had SD corruption issues with my Pis. However, if you want to store the whole media folder  (thumbnails, and separate storage subfolders) on a separate volume you can easily achieve that by linking media from the webfolder to the separate volume.

I'll look into shutter speed but a quick check didn't show a problem but I don't use the HQ camera.

username7291, 2021-01-07:
I everytime watch all the Videos with a Mediaplayer. I would not able to remember witch i have already seen. Cause i delete the seen. Its not a big problem, i have made the media folder available via apache2 for viewing on a device without Samba.

Cant say if its with other Camera Modules, tryed your great Software 2017. But the V2 Camera Module had a video under low light conditions like a smartphone from 2014, not useable. 

There is another issue, maybe related to HQ camera. The Timelapse function, taking a Picture takes alot of time, the stop Button is only available very very short. I googled around, the HQ camera should able to take 4K at 60fps but only with 4 Lanes, means on rpi it should able to take 4K at 30fps but not able to encode. Then i found out taking a picture is slow cause there are lightning measurements are done in front of taking pictures. So ppl should leave the cam running then the measurements are not done again and they get fast pictures. Between the timelapse images, i see sometimes a single frame update in the gui. I quess it switch back to video mode and do the measurements&switching over again. Like if i hit "record image" all 3 seconds.
Wouldnt it better to make a real timelapse like on Photocameras?